openrocket/openrocket #1225: Calculate drag-separation force

neilweinstock, 2022-03-06:
Inspired by a question from user @Kelly on TRF: it would be really neat if, as part of flight simulation, OR could calculate the drag separate force post-burnout.  In order to do this, it would be necessary to be able to specify the separation point of the rocket (or two of them in the case of dual-deploy.)

I'm not sure how it could all work but it would be a very nice feature for HPR fliers.

kejones62, 2022-03-06:
This is Kelly from TRF.  
It's actually not even necessary for the user to specify the separation point (although this would be nice).  OR, I think, already knows the _possible_ points of separation, and could simply calculate the separation force between each set of adjacent components.  So if the model consisted of a nosecone, payload section, and booster, the tool could calculate the drag force between nosecone and payload, and between payload and booster.  If the latter joint is pinned and will not separate, the user can just ignore this value.
What is needed to do this is the mass of each subcomponent (which OR I think already knows), and the drag force on each component.  I'm not sure if OR computes the per-component drag forces independently, but I'm guessing it does.  If so then this should be simple, and I'd be willing to help with the effort on this.

neilweinstock, 2022-03-06:
Well... some rockets may have a number of possible separation points, I'm not sure it's useful to analyze all of them.  But we'll have to think about a smart implementation.

Initially at least, the easiest way to make this information available to the user is as another source of data in simulation plots.  We already have drag force there.  Just need to clarify which separation point is being tracked.

In any case: this will have to wait until after the current release, but then I think it'll be a great thing to dig into.

JoePfeiffer, 2022-03-06:
We only really care about the maximum, so an extra column in the component analysis might make sense.

neilweinstock, 2022-03-06:
Yeah, that would be fine... I imagine, though, that once it's being calculated across the entire flight, then you might as well offer it in the sim plots.  Would be interesting if nothing else.

But yeah, maximum is the number that is most important.