openrocket/openrocket #1226: Not-a-number (NaN) exception when simulating large staged clusters

questionablerocket, 2022-03-07:
Not-a-number (NaN) exception when simulating 15 x D12-0 to 15 x D12-7 staged cluster (originally encountered with A10s but reproduced with D12s to avoid any complications caused by my workaround of issue #1219). The rocket has 3 clusters of 5 motors each in each stage, since there is no option for 15 motor clusters otherwise. Simulations work for the sustainer alone but fail with an internal exception error when a booster is added:

    ERROR (SwingExceptionHandler.java:119) An exception occurred during the simulation
    > net.sf.openrocket.util.BugException: BUG: Simulation resulted in not-a-number (NaN) value, please report a bug, c=(NaN,NaN,NaN)

I'm not running the beta, but I am building unstable every time I see a new commit.  Builds that produced the bug report below were from 3 March at 18:36 UTC, right after pull request 1213 was merged, though the problem persists after more recent commits.

Platforms are:
        Ubuntu 18.04 with the 5.4.0-99-generic kernel running on an octo-core, 16-thread Xeon E-2286M with 128 GB RAM
        MacOS 12.2.1 running on a base model M1 Air with 8 GB RAM.

The simulation will run in 15.03, albeit with questionable plots and extraordinarily long run times (26 seconds on Linux/Xeon, 27 seconds on MacOS/M1, compared to nearly instantaneously for most simulations on either platform), but it fails on both platforms in unstable with either the 15.03 ork or one made using unstable (parts are slightly different due to the different databases). Changing ignition on the sustainer from automatic to burnout does not help.

The ork files created in 15.03 and unstable can be found in this post on TRF since I can't figure out how to attach them here.

https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-is-underway-download-it-now.171295/post-2249083

This is an edge case with an obvious workaround of making custom engine files aggregating motors, so it's not necessarily a high priority unless indiciative of a deeper problem.

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.
Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation
at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120)
at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:414)
at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:98)
at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750)
at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847)
at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112)
at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857)
at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317)
at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249)
at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
at java.desktop/java.awt.Dialog.show(Dialog.java:1070)
at java.desktop/java.awt.Component.show(Component.java:1716)
at java.desktop/java.awt.Component.setVisible(Component.java:1663)
at java.desktop/java.awt.Window.setVisible(Window.java:1031)
at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005)
at net.sf.openrocket.gui.main.SimulationPanel$3.actionPerformed(SimulationPanel.java:177)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635)
at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
[... 282 lines cut ...]
19529 48.069 VBOSE (BasicEventSimulationEngine.java:265) Obtained event from queue: FlightEvent[type=ALTITUDE,time=1.65,source=Rocket,data=[37.62958786889132;39.22478996610364]]
19530 48.069 VBOSE (BasicEventSimulationEngine.java:266) Remaining EventQueue = [FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12], FlightEvent[type=BURNOUT,time=3.3,source=Inner Tube,data=D12]]
19531 48.069 VBOSE (BasicEventSimulationEngine.java:348) Handling event FlightEvent[type=ALTITUDE,time=1.65,source=Rocket,data=[37.62958786889132;39.22478996610364]]
19532 48.069 VBOSE (BasicEventSimulationEngine.java:137) Taking simulation step at t=1.65 altitude 39.22478996610364
19533 48.103 ERROR (SwingExceptionHandler.java:119) An exception occurred during the simulation
> net.sf.openrocket.util.BugException: BUG: Simulation resulted in not-a-number (NaN) value, please report a bug, c=(NaN,NaN,NaN)
> at net.sf.openrocket.simulation.AbstractSimulationStepper.checkNaN(AbstractSimulationStepper.java:219)
> at net.sf.openrocket.simulation.RK4SimulationStepper.computeParameters(RK4SimulationStepper.java:299)
> at net.sf.openrocket.simulation.RK4SimulationStepper.step(RK4SimulationStepper.java:124)
> at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulateLoop(BasicEventSimulationEngine.java:138)
> at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulate(BasicEventSimulationEngine.java:90)
> at net.sf.openrocket.document.Simulation.simulate(Simulation.java:373)
> at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:55)
> at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:24)
> at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
> at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
> at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
> at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
> at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
> at java.base/java.lang.Thread.run(Thread.java:829)
19534 48.103 INFO (SwingExceptionHandler.java:142) Running in EDT, showing dialog
19535 48.103 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
19536 49.432 USER (SwingExceptionHandler.java:220) User requested sending bug report
19537 49.437 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

questionablerocket, 2022-03-07:
Formattiing of the above report is wonky due to my forgetting how to paste code blocks into github.  The bug report can also be found in the aforementioned TRF post.

JoePfeiffer, 2022-03-07:
You can attach them to issues and comments in github by changing the file extension. So call it rocket.ork.txt or some such and github will take it (if anyone can find a setting anywhere in github that will let it accept custom file extensions, we will be eternally grateful).

PR #1236 (2022-03-14): Check for active upper stage before dropping booster
If there are multiple independent motor clusters in a stage, each of them can attempt a separation event. When the second event happens, there is no longer an active upper stage, so the separation results in a simulation branch with no active stages.  This causes a NaN exception as the mass is 0.

fixes #1226
closed 2022-03-14 by SiboVG