openrocket/openrocket #1227: When exporting templates, draw fin roots correctly for fins-on-nose cones and fins-on-transitions

neilweinstock, 2022-03-07:
As reported by @MadRocketer on TRF: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-is-underway-download-it-now.171295/post-2250080

When a fin is placed on a transition or nose cone, the exported template still shows a straight edge.
[image]
Exports like this:
[image]

Obviously it would be preferable for the exported template to show the correct root edge contour.

hcraigmiller, 2022-03-07:
This may be associated with Issue #1021 "Bug: Fin root edge not drawn correctly on transitions". Although Issue #1021 references only transitions specifically, it applies to nose cones as well.

neilweinstock, 2022-03-07:
Related but not the same.

SiboVG, 2022-08-14:
Note: the issue is not only present in the exported template, in certain scenarios you can see the fin protruding the inside of the transition.
[image]

PR #1598 (2022-08-23): [#1227] Fix fin root edge display in 3D view and fin template
This PR fixes #1227. The root edge of a fin set now correctly outlines a transition's shape both in the 3D view and in the fin template export. Besides the fins, the fin tabs also weren't drawn correctly (neither in the 2D view, nor in the 3D view - the fin tab root edge was always just a straight line). This PR solves that issue as well.

Demo (with a transparent transition):

https://user-images.githubusercontent.com/11031519/185255413-d52b3dcc-c1fa-4c30-9025-a2d433d4c4f4.mp4

Fin template export also works with fin tabs:
[image]
closed 2022-08-23 by SiboVG