openrocket/openrocket #1231: "fit" doesn't always work in side view

sjh114961, 2022-03-07:
"fit" doesn't always work in side view. The ends sometimes go off screen and you can pan over to see end. See exampl
[sjh_trident.zip](https://github.com/openrocket/openrocket/files/8200848/sjh_trident.zip)
e

hcraigmiller, 2022-03-08:
The problem presented is that _Inner Components_ which extend beyond the "airframe" are not included in the length of the rocket for the purpose of calculating the overall rocket length (shown in the upper left of the display window), determining the size of the display window, or otherwise.

The collateral issue raised by this issue is how to adjust the flight characteristics computation to take into account inner components extending into the airflow.

JoePfeiffer, 2022-03-08:
The basic problem here is that this model has an "inner" tube that's partially external.  The best fix for it would be to replace it with an inner tube that reached to the end of the body tube, and then a body tube with the same diameter extending from there.

Modifying the "fit" function so it looked at all components, internal and external, for its calculations would be reasonable. I *really* don't want to get in to inferring what parts are and are not exposed to the airflow for simulations.

hcraigmiller, 2022-03-08:
>Modifying the "fit" function so it looked at all components, internal and external, for its calculations would be reasonable.

I too think this is the best approach. Those wanting airflow analysis can use _Body Components_ instead of _Inner Components_. Kind of the best of both worlds.

sjh114961, 2022-03-28:
This still seems to be a problem in beta 2

neilweinstock, 2022-03-28:
> Modifying the "fit" function so it looked at all components, internal and external, for its calculations would be reasonable.

I vote a hearty "aye" to this.  I'm a bit confused why it would do anything else?

PR #1417 (2022-06-09): [#1231] Include non-aerodynamic components in bounding box calculations
This PR fixes #1231 by including non-aerodynamic components in the rocket's bounding box calculations. 

I first just deleted the aerodynamic condition in the bounding box calculations, but most parts of the program (including some calculations) need the bounding box or rocket length of only aerodynamic components. Just deleting the aerodynamic condition caused changes in the calculations.

So I ended up splitting the bounding box and rocket length in two variations: only aerodynamic components and all components.

This PR [... cut]
closed 2022-06-09 by SiboVG