openrocket/openrocket #1244: [Bug] Application of selected freeform fin color is intermittent.

hcraigmiller, 2022-03-14:
Whether both sides are the same selected color, or different. the selection is intermittently applied in the 3D views. See attached .ork file

[X-15.Rev.004.ork.txt](https://github.com/openrocket/openrocket/files/8242287/X-15.Rev.004.ork.txt)

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]

hcraigmiller, 2022-04-03:
I've been able to reliably create this error using this process:

1. Open the _A simple model rocket_ example.
2. Select 3D Finished view.
3. Delete the fins.
4. Add a trapezoidal fin set (use the default settings).
5. Select _Trapezoidal Fin Set_ &rarr;  _Trapezoidal Fin Set configuration_ &rarr; _Appearance,_ then uncheck the default appearance box to verify the fin appearance changes.
6. Select _Trapezoidal Fin Set_ &rarr;  _Trapezoidal Fin Set configuration_ &rarr; _General_, then split the fins.
7. Select _Trapezoidal Fin Set No.1_ &rarr;  _Trapezoidal Fin Set configuration_ &rarr; _Appearance,_. The default appearance is unchecked and the fin color selected is black (no longer white), but the fins are still wood grain (the default appearance); unchecking and rechecking the default appearance box has no effect.

[image]

8. Now, save the design, and reopen the file.
9. Select _Trapezoidal Fin Set_ &rarr;  _Trapezoidal Fin Set configuration_ &rarr; _Appearance,_. The texture now shows the wood grain .jpg file, which was absent in Step 7.

[image]

10. Recreate all steps through Step 7.
11. Then, select _Trapezoidal Fin Set_ &rarr;  _Trapezoidal Fin Set configuration_ &rarr; _Appearance,_. Now, select one of the decals as the texture (initially selecting _none_ has no effect), then select _none_ for the texture.

[image]

Based upon the above, it appears that when the fins are split that (in addition to not preserving the fin color selected), **the texture .jpg is applied, but not shown as the texture**.

[Windows 11 Pro, Version 21H2, Build 22000.434, Windows Feature Experience Pack 1000.22000.434.0]
[Java 17]

SiboVG, 2022-06-13:
@hcraigmiller are you still experiencing this error? Can't seem to reproduce it following your steps

hcraigmiller, 2022-06-13:
Yes, but I added a step that was missing... in bold above.

SiboVG, 2022-06-13:
Ah no, I got it, in step 5, the box should be checked (use default), in that case I can recreate the issue