openrocket/openrocket #1245: I/O error while saving model

sjh114961, 2022-03-14:
I am trying save a mode and getting I/O error. See included screenshot.
[image]

hcraigmiller, 2022-03-15:
More information is likely going to be needed. Can you provide the .ork design file in blocks that can be saved, that can be assembled to create the design that you cannot save, and/or can you provide a bug file?

sjh114961, 2022-03-15:
Well, since it gave me an error trying to save it, no. No bug file as far as I know was created.

hcraigmiller, 2022-03-15:
This appears to be the Black Star Voyager 5 design for which the .ork file was posted on the OpenRocket Facebook page about five hours ago, with the tag line (by @sjh114961 ): "My latest creation. This one was extremely taxing, took me about a week to figure how to do the triangular fins."

In light of the lack of information provided by the user to enable this Issue to be analyzed and resolved, and, absent further information from the User, I recommend that this issue be closed.

neilweinstock, 2022-03-15:
I'm going to have to have a look at this one.  It's not clear to me why all the pods were necessary to do this rocket; in fact I suspect the above model could have been created with zero use of pods.  But I'll need to look more closely.

> That said, the design does produce a side view that bears some looking in to.

Both the side view and the 3D view still do not calculate extents correctly to include all pod and internal elements.  I have more issues to file once I do a little more work to isolate the issue.

> That said, this appears to be the type of design that would benefit from inner body tube extensions such as those described by @neilweinstock in his new feature request Issue #1056.

There are many many many designs that would benefit from that feature.

hcraigmiller, 2022-03-15:
If you want to play with this one... here's one with a corrected airframe {no pods, no discontinuities].

[image]

[BlackStarVoyager5.Deconstruct.Airframe.ork.txt](https://github.com/openrocket/openrocket/files/8253924/BlackStarVoyager5.Deconstruct.Airframe.ork.txt)
[BlackStarVoyager5.Deconstruct.Airframe.Fins.ork.txt](https://github.com/openrocket/openrocket/files/8253931/BlackStarVoyager5.Deconstruct.Airframe.Fins.ork.txt)

Have fun...

sjh114961, 2022-03-16:
I got one that works, thanks.

neilweinstock, 2022-03-18:
Closing this.

closed 2022-03-18 by neilweinstock