openrocket/openrocket #1257: [Bug] can't read selected rows in tables on Linux

cabbey, 2022-03-22:
I'm running on Ubuntu Linux 21.10.

Anywhere in the UI that uses a table is rendering with the selected cell/row of the table in black text on a black background. A similar color contrast issue is present when rendering dropdowns. This can be seen with both the 22.02.beta.01 installer delivered file and with a locally built jar from the unstable branch at 0adcf040ba84059a831cc4d2ad3426a977f0df91

[image]

[image]

[image]

---------- System information ----------
OpenRocket version: 22.02.beta.01
OpenRocket source: default
OpenRocket location: /home/cabbey/OpenRocket/OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.X11.XToolkit
exe4j.moduleName=/home/cabbey/OpenRocket/OpenRocket
file.encoding=UTF-8
file.separator=/
install4j.appDir=/home/cabbey/OpenRocket/
install4j.exeDir=/home/cabbey/OpenRocket/
install4j.jvmDir=/home/cabbey/OpenRocket/jre
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.X11GraphicsEnvironment
java.awt.printerjob=sun.print.PSPrinterJob
java.class.path=/home/cabbey/OpenRocket/.install4j/i4jruntime.jar:/home/cabbey/OpenRocket/.install4j/launcher47cd7162.jar:/home/cabbey/OpenRocket/OpenRocket.jar
java.class.version=55.0
java.home=/home/cabbey/OpenRocket/jre
java.io.tmpdir=/tmp
java.library.path=/usr/java/packages/lib:/usr/lib64:/lib64:/lib:/usr/lib
java.net.useSystemProxies=true
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.14.1+1-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.14.1
java.version.date=2022-02-08
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.14.1+1-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln3240169098456319610
line.separator=\u000a
os.arch=amd64
os.name=Linux
os.version=5.13.0-37-generic
path.separator=:
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
sun.boot.library.path=/home/cabbey/OpenRocket/jre/lib
sun.cpu.endian=little
sun.cpu.isalist=
sun.desktop=gnome
sun.font.fontmanager=sun.awt.X11FontManager
sun.io.unicode.encoding=UnicodeLittle
sun.java.command=install4j.net.sf.openrocket.startup.OpenRocket
[... 118 lines cut ...]
271 7.788 VBOSE (Rocket.java:440) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@7db7dad3, current number of listeners is 23
272 7.789 VBOSE (Rocket.java:440) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@54cd450a, current number of listeners is 24
273 7.790 VBOSE (DoubleModel.java:897) DoubleModel[Rocket:OverrideCD] adding listener (total 1): javax.swing.JSpinner$ModelListener@bf4159b
274 7.791 VBOSE (DoubleModel.java:897) DoubleModel[constant=1.0] adding listener (total 1): net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@79e4342d
275 7.791 VBOSE (DoubleModel.java:897) DoubleModel[Rocket:OverrideCD] adding listener (total 2): javax.swing.JSlider$ModelListener@473fcb0e
276 7.791 VBOSE (DoubleModel.java:897) DoubleModel[Rocket:OverrideCD] adding listener (total 3): javax.swing.plaf.basic.BasicSliderUI$Handler@6ccc56bf
277 7.791 VBOSE (DoubleModel.java:919) DoubleModel[Rocket:OverrideCD] removing listener (total 2): javax.swing.plaf.basic.BasicSliderUI$Handler@6ccc56bf
278 7.791 VBOSE (DoubleModel.java:897) DoubleModel[Rocket:OverrideCD] adding listener (total 3): javax.swing.plaf.basic.BasicSliderUI$Handler@155831c5
279 7.791 VBOSE (Rocket.java:440) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@749bff02, current number of listeners is 25
280 7.791 VBOSE (Rocket.java:440) Added ComponentChangeListener net.sf.openrocket.util.ComponentChangeAdapter@749bff02, current number of listeners is 25
281 7.809 INFO (OpenRocketDocument.java:536) Adding undo position 'Modify Rocket' to net.sf.openrocket.document.OpenRocketDocument@33c5b84a, document was in clean state
282 7.877 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
283 7.877 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
284 7.878 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
285 7.880 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
286 7.881 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
287 7.881 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
288 7.919 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
289 7.920 VBOSE (AbstractScaleFigure.java:191) ____ Updating RocketFigure to: 0.562712 user scale, 2126.79 overall scale
290 10.488 DEBUG (GUIUtil.java:232) Clearing all models of window net.sf.openrocket.gui.configdialog.ComponentConfigDialog[dialog0,1525,549,555x462,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title= Rocket configuration,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,37,555x425,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@41d9d0a9,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
291 10.491 DEBUG (MemoryManagement.java:56) Adding object into collectable list: net.sf.openrocket.gui.configdialog.ComponentConfigDialog[dialog0,1525,549,555x462,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title= Rocket configuration,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,37,555x425,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@41d9d0a9,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
292 10.492 VBOSE (Rocket.java:446) Removed ComponentChangeListener net.sf.openrocket.gui.configdialog.ComponentConfigDialog[dialog0,1525,549,555x462,invalid,hidden,layout=java.awt.BorderLayout,MODELESS,title= Rocket configuration,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,37,555x425,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=javax.swing.plaf.synth.SynthBorder@41d9d0a9,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true], current number of listeners is 24
293 17.130 USER (BasicFrame.java:822) Bug report selected
294 17.134 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2022-03-23:
Do you have any other desktops available to you?  I've been doing all of my development under Debian, with the xfce desktop, and have not observed this.  Installing gnome now...

cabbey, 2022-03-23:
I've just noticed that it's NOT present with the OS installed JVM, only with the one the installer choose to install:

works fine:
```
cabbey@cheshire:~$ java --version
openjdk 11.0.14 2022-01-18
OpenJDK Runtime Environment (build 11.0.14+9-Ubuntu-0ubuntu2.22.10)
OpenJDK 64-Bit Server VM (build 11.0.14+9-Ubuntu-0ubuntu2.22.10, mixed mode, sharing)
```
shows the problem:
```
cabbey@cheshire:~$ OpenRocket/jre/bin/java --version
openjdk 11.0.14.1 2022-02-08 LTS
OpenJDK Runtime Environment (build 11.0.14.1+1-LTS)
OpenJDK 64-Bit Server VM (build 11.0.14.1+1-LTS, mixed mode)
```

cabbey, 2022-03-25:
@JoePfeiffer as you mentioned on slack,  I'm suspecting it's something about the modifications Ubuntu or Debian are making to the upstream build. I've been poking at what all they change and, well, it's a lot, and nothing seems to jump out as the key change. http://archive.ubuntu.com/ubuntu/pool/main/o/openjdk-lts/openjdk-lts_11.0.14+9-0ubuntu2~22.10.debian.tar.xz

One of the big things they do is step the default gtk back to version 2 instead of version 3... playing with that I stumbled over an interesting observation... when it is legible, I get this error logged from the jvm. (note, I get this error with *all* java apps that use swing.)

```
Gtk-Message: 20:42:32.091: Failed to load module "canberra-gtk-module"
```
if I don't get that message, either because I use gtk3 with the installed jvm, or because I use any version of gtk with the bundled jvm, then I can't see the text. Sadly that error message doesn't seem to lead to any of the changes they are making to the jvm.

jimmiedave, 2022-04-18:
Checked a minute ago: I'm on Ubuntu 20.04 LTS, and have no legibility problem, but notice that Ubuntu and OpenRocket appear to have the same (or very close) JVM versions  installed:

```
~$ java --version
openjdk 11.0.14.1 2022-02-08
OpenJDK Runtime Environment (build 11.0.14.1+1-Ubuntu-0ubuntu1.20.04)
OpenJDK 64-Bit Server VM (build 11.0.14.1+1-Ubuntu-0ubuntu1.20.04, mixed mode, sharing)

```
```
~$ OpenRocket/jre/bin/java --version
openjdk 11.0.14.1 2022-02-08 LTS
OpenJDK Runtime Environment (build 11.0.14.1+1-LTS)
OpenJDK 64-Bit Server VM (build 11.0.14.1+1-LTS, mixed mode)

```

SiboVG, 2024-12-26:
This should be fixed after the switch from Linux's native LAF to FlatLaf.

closed 2024-12-26 by SiboVG