openrocket/openrocket #1264: resizing

gri2a, 2022-03-25:
In 15.03, when I build a rocket I start with body tube. Default is 50mm diameter. I change this to (eg) 76mm
I then add a nose cone. It creates a 50mm nosecone behind the body tube. I move the Nose Cone to the front of the body tube and it auto resizes to 76mm.
In 22.02, when I relocate the nosecone to the front of the body tube it does NOT resize automatically.
Only a small issue, but a retrograde step none the less.

JoePfeiffer, 2022-03-25:
It looks like 15.03 created new components with the "Automatic" size flag checked; 22 doesn't.  Oddly, I wasn't even able to check it for a nose cone until after I moved it forward.

SiboVG, 2022-03-25:
The disabled checkbox is intended behavior, it will only become enabled when there is a body tube or transition after the nose cone.

SiboVG, 2022-03-25:
The 'automatic'  checkbox being unchecked is also intended behavior. When you insert a nose cone and there isn't a component that the nose cone can take its dimensions from, the automatic checkbox is disabled and unchecked.

JoePfeiffer, 2022-03-25:
If it's all expected behavior, do we close this?  It seems like letting a nose cone grab its radius from a preceding body tube if there's no trailing one (which seems to be what 15.03 did) would be reasonable.

hcraigmiller, 2022-03-25:
I think it should be closed. If a user decides to build a rocket in an unconventional order, that is not a bug. The way that it currently functions seems like the best approach overall.

closed 2022-03-25 by SiboVG

hcraigmiller, 2022-03-25:
Whatever decision is made, the nose cones and transitions should behave the same.

hcraigmiller, 2022-03-25:
There may be a better approach to this issue, however.

Currently (starting with a new design), add a body tube and the diameter is the system default value that I’ll call 2, with the automatic box greyed out: **change the 2 to 1**. Now, add a second body tube after the first, and that body tube has the automatic box checked and adopts the diameter of 1, the new diameter of the preceding body tube (the desired behavior). But, **if you uncheck the automatic box of second body tube, the diameter reverts to the system default of 2** (most likely not the desired behavior).

**The better behavior would be to have the diameter of a component adopt its then present diameter as its new default when the automatic box in unchecked**. So in the example above, when you uncheck the automatic box of second body tube, the diameter would remain at 1 (most likely not the desired behavior). 

In this way, as an example _with the suggestion of @JoePfeiffer above implemented_, with the automatic boxes checked on a transition (with several sizes of body tube above and below), as you move the transition up and down the airframe looking at how different placements look with different diameter tubes, when you uncheck one of the automatic boxes it adopts the shown diameter as its new default for that end, rather than reverting to the system default and forcing you to manually enter the desired diameter before you can continue looking at diameters for the other end.