openrocket/openrocket #1265: Report CP and stability from different axes

neilweinstock, 2022-03-25:
Asymmetrical rockets will have a CP and stability margin that varies with the position of the rocket.  OR should report this in some way.

Best UI to do this is TBD.

JoePfeiffer, 2022-03-25:
Thinking about it, since OR actually calculates CP in all roll angles, it wouldn't be too hard to add. One thing I wouldn't want is to just display the CP in a given orientation; it would be too easy for a rocketeer, out of either ignorance or malice, to show an RSO an OR design carefully oriented to look stable and have the RSO not have any better response than "it don't look stable to me and I say no". Maybe if the rocket stats in the upper right hand corner showed a CP range for a rocket that was asymmetric enough to matter? And then have the CP icon in the figure move as you changed the roll angle? Hmmm....

JoePfeiffer, 2022-03-25:
I had actually thought the moving CP icon was already implemented -- good thing I checked before responding on TRF!

jlassahn, 2022-03-26:
When I fantasize about this feature, I imagine a dot for the minimum CP, another dot for the maximum, and a line between them.  Maybe also the min and max values reported as numbers.

neilweinstock, 2022-03-26:
There are so many potentially interesting ways to visualize this.

A polar plot of stability, with a "unit circle" around it, would be cool (suggest by Joe Avins on TRF).  We should see exactly how Rocksim reports it, for reference.

Updating the CP position and stability margin in the rocket display as the side view is rotated is probably good... but I'm not sure if there are times that you *wouldn't* want it to do that.

Most likely something to experiment with a bit to see what type of visualization is most effective.