openrocket/openrocket #1269: Windows 10 install fail.

sjh114961, 2022-03-26:
When installing beta 2 screen freezes on install screen. See screen shot. I reported this same problem on the beta 1. Fortunately, the code is fully installed at this point.  I waited 15 minutes and neither the "X" in upper or clicking the "Cancel" have no effect. I had to go into the windows Task manager to finally get rid of the screen.
[image]

SiboVG, 2022-03-27:
Does this always happen or were you eventually able to correctly install the software?

@hcraigmiller did you ever experience this?

JoePfeiffer, 2022-03-27:
He reports in FB that also overwrote beta 1 in spite of checking the box saying not to.

sjh114961, 2022-03-27:
The software does appear to install. Just the installer gets stuck and I have to use the Task Manager to kill it. I did report the same issue in Beta 1, but I can't seem to find it

JoePfeiffer, 2022-03-27:
I wonder if there's a way to get an install log?  That might have a clue...

hcraigmiller, 2022-03-27:
> Does this always happen or were you eventually able to correctly install the software?
> 
> @hcraigmiller did you ever experience this?

No, not at all.

sjh114961, 2022-03-27:
If you guys have a debug version with some sort logging or debug info. I would be glad to try again.

JoePfeiffer, 2022-03-27:
I think I've found a way; it can be done with command-line options to the standard installer you've already got and doesn't need a special debug installer.

Bring up a command prompt window, cd to the folder where you put the installer, and give the
command
OpenRocket-22.02.beta.02-Windows.exe -Dinstall4j.debug=true -Dinstall4j.keepLog=true

Give it a good long time to finish and then kill it as you've done before.  There should now be a file called:
C:\Program Files\OpenRocket\.install4j\installation.log
Please upload that here if it'll let you, or else email me a copy.

sjh114961, 2022-03-28:
[installation.log](https://github.com/openrocket/openrocket/files/8359126/installation.log)
Here it is.

JoePfeiffer, 2022-03-28:
I'm very sorry to say I'm at a loss.  I've compared your log to my own
[my-log.log](https://github.com/openrocket/openrocket/files/8363726/my-log.log)
and I'm not seeing any significant difference at the end that would explain why my laptop brings up the Finish dialog and yours doesn't.  Hoping somebody more Windows-savvy than me picks this up... (puzzling that it looks like yours is the log from back when you installed beta 1 a month ago, but that shouldn't be important).

sjh114961, 2022-03-28:
I am at a loss why the date on the log is a month old. Since you said it matched yours, I decided to disable my virus protection (Trend Micro) and run it again. This time everything worked properly. So it looks like the virus protection was the issue.

JoePfeiffer, 2022-03-28:
That you found a solution is terrific news!

closed 2022-04-02 by JoePfeiffer