openrocket/openrocket #1270: Uncaught exception when attempting to add custom script to simulation.

Sparrowhawk120, 2022-03-28:
**STEPS TO REPRODUCE: **
Open any rocket file and create a simulation. Edit the simulation, and go to "Simulation Options". Select "Add Extension", "User Code", "Scripts". The error is thrown.

The generated bug report has been attached [here](https://github.com/openrocket/openrocket/files/8365150/Bug.Report.txt).

Are these custom script options even used for anything?

SiboVG, 2022-03-28:
Running OR from a later version (I tested it with Java 17) resolves the issue. Maybe we should consider using a later Java JRE version for OR?

Note: you cannot run OR with a JRE other than Java 17 because of the restriction in the latest beta, I just edited some code to allow me to change the version.

hcraigmiller, 2022-03-28:
At some point we need in the near future development needs to shift from Java 11 to Java 17. The question is whether to make that shift before or after the next "release". Does anyone know what kind of potential issues would occur if the shift were made now, or does someone need to test that out?

JoePfeiffer, 2022-03-28:
The best person to answer this is @thzero -- how close is #1222 to being ready?

JoePfeiffer, 2022-03-28:
Is this the same as #1108?

Sparrowhawk120, 2022-03-29:
It seems pretty similar, they're both uncaught exceptions on scripting.

thzero, 2022-03-29:
Well I think all the dependencies work, and I can't say I did vast testing but outside of the scripting which I can't seem to get to update things seemed to work on java 11.   Didn't try java 17.

Problem with scripting is Oracle has been deprecating things left and right, and the scripting is one including the javascript engines - and no one is really seemingly interested in filling in the void.

Kinda same with the 3d libraries from swing.

thzero, 2022-03-29:
If you want to try the dependencies, I'd create another branch and bring them in, and test.

SiboVG, 2022-03-29:
> Is this the same as #1108?

Not really, #1108 still persisted even in Java 17.

SiboVG, 2022-04-08:
I suggest we close this issue, as we know it is solved by updating to Java 17

SiboVG, 2022-04-08:
My apologies, forgot that the issues still persists when you run OR from the JAR, the issue was just resolved when I run OR from my IDE in Java 17

thzero, 2022-04-08:
Correct.  It works in the IDE with Java 11 too.   Issue seems to be the use of the class loader and the scripting engine doesn't get included when executing when executing from the .jar

thzero, 2022-06-26:
So been looking at this again.   Its definitely the do the JarInJar and using a customer classloader which seems to have been borked in Java 9+.

Launching from SwingStarter the Javascript works, but the external jar plugins don't.
Launching from OpenRocket the external jar plugins work, but the Javascript plugins give the class not found when trying to load the ScriptingEngineManager.    Now the system class loader instead of being a URLClassloader is an internal class of the variety of ClassLoaders.AppClassLoader (mega useful).

In the latter case, its you can see the script-api dependency in the ucp of the classloader, but something is missing.

I've tried to use the javax.instrumentation as a work around to load the external jar based plugins up.   Seems to be able to create a class through reflection, but not able to discover it and instantiate with Guice.   Still looking at this as it may hold some promise.

I've also tried a couple of variations on creating new ClassLoaders; most of these have involved creating a new URLClassLoader and loading the plugins and then using the SwingStarter.class.getClassLoader() as the parent.  This allows the scripting engine to work, but then the external plugins are not available.
Also tried a custom class that extends from URLClassLoader.    This one makes it a bit easier to see what track of whether in a given situation the thread or class is using the expected class loader.   Nonetheless, same scenario here.

Loading the external plugin jar on the classpath does seem to work.  That may be the required work around in order to have both features.

This is all on Java 11 only.

thzero, 2022-06-27:
Ah... I have made progress!   
This a hack I feel - I still don't quite understand what is missing when setting up the URLClassLoader that the scripting engine is not found.   However, combination of abusing the javax.instrumentation functionality to add a new class into the system classloader so now the class is in the loader for use by Guice.   However, the annotation finder defaults, as the classloader is not a urlcassloader, and therefore the annotation finder only searches the classpath.  These plugins are not on the classpath, but when it seems you can add them to the classpath and the annotation finder finds them.   Then boom both the scripting and the external java based plugins are avaiable (or at least seem to be).

thzero, 2022-06-28:
Well, with everything, there is always a ying and a yang.

This works ok it seems (not even dealt with the unit test yet) in IntelliJ IDE, and then tried to test it through the build scripts and running the jar.

So the ramifications of this are

- The jar can not be run naked, i.e. java -jar <jar name>
- There are two issues that cause this
  - First is the instrumentation Agent.  It requires a command line switch in order to execute.  Command line is java -javaagent:./OpenRocket.jar (I just added the Agent class into the standard OR jar).
  - Second is SwingStartup
    - At least in Java 11, the gluegen-rt.jar and jogl-all.jar are not referenced from within the jar, they have to be externalized.
    - The Classpath-Jars does not work, or was only used by the JarInJar code to get additional jars and put them on the classpath.
    - Externalizing these jars into a libs folder along side the OpenRocket.jar, allows the application to boot up.   Otherwise it will crash as it cannot find certain gluegen classes.   
    - This requires another command line directive, -cp .:lib/*
 - So end of the day, now the launch sequence for this approach is as follows
   - java -cp .:lib/* -javaagent:./OpenRocket.jar -jar OpenRocket.jar
   - Requires the gluegen-rt (and natives) and jog-all (and natives) jar in a libs folder in same location as OpenRocket.jar

thzero, 2022-06-28:
Really the JarInJar just seems to be a complete hack at this point.
I've also investigating module layers as another potential to add java plugins in with SwingStarter, but that would still come up with the issue of the gluegen/jogl libs.

SiboVG, 2022-06-28:
> Why is ServiceConfigurationError missing? Not sure, can't seem to find it in classloaders so must be a difference in the boot class loader.

sun.misc.SerciveConfigurationError is from Java 8, and has been removed since Java 9, and replaced by java.util.ServiceConfigurationError. So the ScriptEngineManager is references the from ScriptEngineManager version (it loos like it's referencing ScriptEngineManager from Java 8 instead of Java 11, which OR runs on).

thzero, 2022-06-28:
It's also confusing that when running without the JIJ, the code runs fine even with the ServiceConfigurationError from sun.misc referenced.   So what is throwing the exception that is then tripping the catch blocks that seem to be bubbling that up.

SiboVG, 2022-06-28:
I did some testing around this issue some time ago and remember that there is a library, `core/lib/script-api-1.0.jar` that contains the ScriptEngineManager, but **with** the sun.misc dependencies (you can open the JAR in IntelliJ's project tree and open ScriptEngineManager in script-api-1.0.jar/javax.script/ScriptEngineManager and you'll see that it uses sun.misc). I then deleted that library, but then the Guice and Guava lib were complaining, so I then updated those libraries, but was then greeted by a number of other issues.

But: my guess is that Guice and Guava use the script-api-1.0.jar, which overrides Java 11's ScriptEngineManager, so you get the error of the sun.misc class not being found, because it is not part of Java 11.

It's a real shit show...

thzero, 2022-06-28:
Yeah, don't disagree that its certainly is.   

One reason I would like to excise the jakarata bits is to remove dependencies that don't get updated, or change direction, etc. and since it has limited use within the Rocksim import/export its a good candidate.
But maybe guice should be the same.   Is somewhat limited use; what is it bringing OpenRocket that a simpler factory/service lookup (yeah yeah about being anti-pattern, blah, blah) can't do?

thzero, 2022-06-28:
Well, made a copy of the ScriptEngineManager and then updated ScriptingUtil to use a single instance of the new version, and reference the Configurator to use the ScriptingUtil and now the ServiceConfigurationError is gone it seems.   This is using the JIJ pattern only.

however, not able to load the Javascript scripting engine either, so seeing what is up with that now.

SiboVG, 2022-06-28:
> however, not able to load the Javascript scripting engine either, so seeing what is up with that now.

The JavaScript engine has been deprecated in Java 11 as well... I have a branch on my repo that replaces the JavaScript engine with Graal.js (which is equivalent to the deprecated JavaScript engine): https://github.com/SiboVG/openrocket/tree/issue-826. More specifically it's these two commits that replace the engine: [commit 1](https://github.com/openrocket/openrocket/commit/a259f2a8bc3f0d2d865a6c24013dd11561f71e45), [commit 2](https://github.com/openrocket/openrocket/commit/bc37dcfea73aaf53c0e8613da495157d43916c08).

thzero, 2022-06-28:
Yes, its been deprecated... but it still works.   Why it cannot be found with the JarInJar solution is the question.   But maybe just going with the Graal.js is worth a shot.

thzero, 2022-06-28:
Well,

using JIJ with a custom ScriptEngineManager (gets rid of that above CNF exception), and the Graal.js act least lets me do

- show a list of script engines (Javascript)
- create and store a Javascript snippet
  - sample was 
`var hello = 'world';`
- execute the Javascript snipet

however, its coming up with this stack trace during the evaluation of the Javascript code

java.lang.NoClassDefFoundError: com/ibm/icu/text/DateFormat
org.graalvm.polyglot.PolyglotException: java.lang.NoClassDefFoundError: com/ibm/icu/text/DateFormat
	at com.oracle.truffle.js.runtime.JSContextOptions.<clinit>(JSContextOptions.java:457)
	at com.oracle.truffle.js.lang.JavaScriptLanguage.<clinit>(JavaScriptLanguage.java:182)
	at com.oracle.truffle.js.lang.JavaScriptLanguageProvider.create(JavaScriptLanguageProvider.java:51)
	at com.oracle.truffle.polyglot.LanguageCache.loadLanguage(LanguageCache.java:548)
	at com.oracle.truffle.polyglot.PolyglotLanguageInstance.<init>(PolyglotLanguageInstance.java:100)
	at com.oracle.truffle.polyglot.PolyglotLanguage.getInitLanguage(PolyglotLanguage.java:152)
	at com.oracle.truffle.polyglot.PolyglotLanguage.ensureInitialized(PolyglotLanguage.java:142)
	at com.oracle.truffle.polyglot.PolyglotLanguage.getOptionsInternal(PolyglotLanguage.java:134)
	at com.oracle.truffle.polyglot.PolyglotLanguage.getOptionValues(PolyglotLanguage.java:202)
	at com.oracle.truffle.polyglot.PolyglotContextConfig.<init>(PolyglotContextConfig.java:266)
	at com.oracle.truffle.polyglot.PolyglotEngineImpl.createContext(PolyglotEngineImpl.java:1745)
	at com.oracle.truffle.polyglot.PolyglotEngineDispatch.createContext(PolyglotEngineDispatch.java:159)
	at org.graalvm.polyglot.Context$Builder.build(Context.java:1832)
	at com.oracle.truffle.js.scriptengine.GraalJSScriptEngine.createDefaultContext(GraalJSScriptEngine.java:342)
	at com.oracle.truffle.js.scriptengine.GraalJSBindings.initContext(GraalJSBindings.java:91)
	at com.oracle.truffle.js.scriptengine.GraalJSBindings.requireContext(GraalJSBindings.java:86)
	at com.oracle.truffle.js.scriptengine.GraalJSBindings.entrySet(GraalJSBindings.java:172)
	at java.base/java.util.AbstractMap.containsKey(AbstractMap.java:144)
	at javax.script.SimpleScriptContext.getAttribute(SimpleScriptContext.java:124)
	at com.oracle.truffle.js.scriptengine.GraalJSScriptEngine.createSource(GraalJSScriptEngine.java:439)
	at com.oracle.truffle.js.scriptengine.GraalJSScriptEngine.eval(GraalJSScriptEngine.java:435)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:247)
	at net.sf.openrocket.simulation.extension.impl.ScriptingExtension.getListener(ScriptingExtension.java:103)
	at net.sf.openrocket.simulation.extension.impl.ScriptingExtension.initialize(ScriptingExtension.java:64)
	at net.sf.openrocket.document.Simulation.simulate(Simulation.java:366)
	at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:55)
	at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:24)
	at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
	at java.base/java.util.concurrent.FutureTask.run$$$capture(FutureTask.java:264)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java)
	at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: java.lang.ClassNotFoundException: com.ibm.icu.text.DateFormat
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:476)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	... 34 more

thzero, 2022-06-29:
Bingo.

thzero, 2022-06-29:
Ok, seems to work now in IntelliJ IDE.
JIJ and then add to add a boat load of dependencies to get Graal.JS working. :(

graal-sdk-22.1.0.1.jar
icu4j-71.1.jar
js-22.1.0.1.jar
js-scriptengine-22.1.0.1.jar
truffle-api-22.1.0.1.jar

See also https://www.graalvm.org/22.1/reference-manual/js/RunOnJDK/
Oddly I didn't seem to need the tregex.jar, but its a might be.

thzero, 2022-06-29:
Tests run clean.   Runs from a build jar file (no switches required).

I've left it in my fork for the moment... take a look, pull it down.
https://github.com/thzero/openrocket/tree/scripting

SiboVG, 2022-06-30:
> Tests run clean. Runs from a build jar file (no switches required).
> 
> I've left it in my fork for the moment... take a look, pull it down. https://github.com/thzero/openrocket/tree/scripting

First tests look promising, no errors! 😍 Go ahead and make a PR for it, I can do a code check then.

SiboVG, 2022-06-30:
> Ok. I ran it through on latest Ubuntu to from test and actual use, and looked good there too. Don't have a MacOS device, so can't do any sanity testing there.

Well I do and it looked good.

thzero, 2022-07-01:
Ok.  submitted.  Sorry was trying to clean up my forked repo so I can submit clean pull requests... :|

PR #1507 (2022-08-02): Resolving issue #1270, unexpected exception with simulation scripting
This PR fixes #1270, fixes #308, fixes #1108 and fixes #826.

As the Nashorn scripting engine is deprecated, as part of the resolution this replaces Nashorn with the Graal.Js scripting engine.  This includes the addition of the following dependencies

graal-sdk-22.1.0.1.jar
icu4j-71.1.jar
js-22.1.0.1.jar
js-scriptengine-22.1.0.1.jar
truffle-api-22.1.0.1.jar
closed 2022-08-02 by SiboVG