openrocket/openrocket #1271: [Bug/FYI] Screen layering when application opens

hcraigmiller, 2022-03-28:
An RTF user reported the following:

[image]

This was the fix the user found: "Basically what I did is, I switched the Graphical settings from NVIDIA to Intel Integrated. Seems to do the trick (so far). I will keep testing it, and in case I get the same glitch again, I'll let you know."

neilweinstock, 2022-04-03:
OK, it sounds like it was fixed... maybe a good thing to document for future reference?

closed 2022-04-03 by SiboVG

SiboVG, 2022-04-03:
> OK, it sounds like it was fixed... maybe a good thing to document for future reference?

Documentation fixed [here](http://wiki.openrocket.info/FAQ#Windows_Graphics_Settings) by @hcraigmiller

Reysean05, 2022-06-02:
Same issue, but I dont have integrated graphics to switch to.