openrocket/openrocket #1275: Windows installer for beta two desktop shortcut just re-runs the installer

neilweinstock, 2022-04-01:
Reported on TRF: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-2-is-now-available.171295/post-2261654, and I have also experienced this.

The Beta 2 installer for Windows puts a shortcut on the desktop, but double-clicking that shortcut just re-runs the installer, rather than the app itself.  It is kind of weird.

My machine is Windows 10.

JoePfeiffer, 2022-04-01:
What I experience on Windows is different, but could easily be confused for this: when I download the installer, it is placed on the Desktop.  When I install OR, I don't get a second desktop shortcut; the installer icon, however, is still there (and so if I double-click it I rerun it).

neilweinstock, 2022-04-01:
My downloads all go to the Downloads folder, not the desktop... I will experiment some more.

JoePfeiffer, 2022-04-01:
I just tried downloading to Downloads, and got no Desktop icon for either the installer nor the installed program.

hcraigmiller, 2022-04-01:
My experience is the same as @JoePfeiffer.

closed 2022-05-24 by neilweinstock