openrocket/openrocket #1277: Improper Placement of Fins from a Published .RKT File

silversurfer77, 2022-04-03:
I opened a .RKT file from Loc's website for a 3.1" Iris rocket:
https://cdn.shopify.com/s/files/1/0568/7489/3503/files/YIRIS338.zip?v=1623761980

Their website has two files...one for the 38mm and 54mm MMT's, but both have the same issue:
https://locprecision.com/collections/rockets-3-1-diameter/products/iris

In OR 15.03 it was fine, but in 22.02 beta 01 it places the fins on the nose cone (as expected CP and CG are different) and for some reason the total length is different by .3 inches.

Here are some pics...
https://imgur.com/a/6qhlorG

JoePfeiffer, 2022-04-03:
Beta 02 has some fixes for .rkt file import -- please try with that and see if you still see the issue.

silversurfer77, 2022-04-03:
OK thanks! I didn't realize 02 was out! The file rendered as expected. Much appreciated for the fast reply Joe!

closed 2022-04-03 by SiboVG