openrocket/openrocket #1279: Feature request: customize mass objects for each motor configuration

neilweinstock, 2022-04-05:
As suggested, sort of, on TRF: https://www.rocketryforum.com/threads/openrocket-hacking-workarounds-kludges-cool-stuff.169967/post-2262869

The desire is to have a customized nose weight configuration for different motor configurations, which seems reasonable.

Issue #689 is related: both concern the need to be able to manage mass objects separately from the main rocket design.  I think it's worth some consideration.

questionablerocket, 2022-04-05:
bad_idea from TRF here.  Copying my TRF post for more background:

Thank you, Neil. If no one knew a simpler way to accomplish this, I had intended to submit a feature request to add configurable mass elements to the configurations created in the Motors & Configuration tab, but I didn't want to pile-on during the beta so was going to submit it after release. I've also been trying to learn the code so as to figure out how this could be added, but it's been very slow going so far.

This would go very well with issue [1053](https://github.com/openrocket/openrocket/issues/1053) to add chute release support, at least for my use of the chute release, where on some rockets I use it for larger motors but I don't with smaller motors. My other - main - use case is to change out nose weights to tune stability when flying different motors. But it would also be useful for cameras, trackers, etc., anything that may be on one flight but not the next.

I have been handling this either by overriding mass from one sim to the next - copying down the results so as to retain them when I revert to the original mass, or by keeping two (or more) separate ork files up to date with the same changes. The fake motors are an attempt to be able to manange these different flight configurations in a single file and without all the back and forth.

In the very long run, it would be ideal to be able to create and sim a more modular rocket in the same file, e.g., to fly with an avbay and and extra body tube in one configuration and not in another - or even to fly a sustainer with a booster in one configuration and on its own in another - but I realize that would be a much larger effort than variable mass elements would be (and I'm sure even variable mass elements won't be simple when all is said and done).

neilweinstock, 2022-04-05:
> (and I'm sure even variable mass elements won't be simple when all is said and done).

Indeed it is not.  But it is a worthy idea for us to think about, in conjunction with #689 and #1053.