openrocket/openrocket #1294: Java 17 support

Licone7, 2022-04-13:
The most recent beta of the software only supports up to Java 11, which is many years gone by. The most recent LTS edition of Java is 17. Is there any reason the software is limited to this older version of Java? If not, I think it'd definitely help if newer versions were supported.

SiboVG, 2022-04-13:
We are currently testing Java 17 compatibility with OpenRocket. The reason why we limited it to Java 11 is because we noticed issues with OpenRocket and Java 14 and 15 and therefore decided to play it safe and not risk other JRE-related bugs to pop up.

If all goes well, the next beta release will be launched with Java 17 compatibility.

closed 2022-04-13 by SiboVG

Licone7, 2022-04-13:
Okay, many thanks!