openrocket/openrocket #1301: crash after adding engine

IlgarB, 2022-04-16:
I tried to add engine and openrocket crashed, Still not working completely uninstall (at least I assume I can uninstall but maybe not because after re-installation still it tries adding motor at the starting and get crash) 

I need help

JoePfeiffer, 2022-04-16:
Does this describe what you're seeing:

1) you added a motor to the user motor directory
2) you tried to start OR; a popup appeared saying "Loading motors"
3) a popup appeared with the title "Error upon thrust curve import" with a message including "Two thrust values for single time point" and giving details about exactly where in the motor thrustcurve the error appeared
4) OR remained in this state until you killed it

If so, thanks -- this hadn't actually been reported here before, but will be fixed before the next beta.  In the mean time, you can edit the motor file to remove the error (if it's a commercial motor, it should already be in OR's database so you don't need your motor file).

If something else, please give us a detailed description of exactly what happened.

IlgarB, 2022-04-16:
That's exactly what happened up to 2'th. Error Message is: IllegalArgumentException:MALFORMED. 

I couldnt open openrocket, already tried uninstall and remove all files. I got the same error again on reinstallation

JoePfeiffer, 2022-04-16:
Does it still happen if you remove the motor file you added?  Please upload the motor file.  You'll need to change the extension to .txt to get github to allow you to upload.

IlgarB, 2022-04-17:
> Does it still happen if you remove the motor file you added? Please upload the motor file. You'll need to change the extension to .txt to get github to allow you to upload.

Yes, I removed everything about openrocket but still can't run program.

Also this is the engine i tried to add: [Cesaroni_M2020_lenght.txt](https://github.com/openrocket/openrocket/files/8500845/Cesaroni_M2020_lenght.txt)

Cesaroni M2020 but lenght is different ( The competition we participated in wants us to design with this engine file, so I tried to install it)

IlgarB, 2022-04-17:
[image]

IlgarB, 2022-04-17:
`---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalArgumentException: MALFORMED
	at java.util.zip.ZipCoder.toString(ZipCoder.java:58)
	at java.util.zip.ZipInputStream.readLOC(ZipInputStream.java:300)
	at java.util.zip.ZipInputStream.getNextEntry(ZipInputStream.java:122)
	at net.sf.openrocket.file.motor.ZipFileMotorLoader.load(ZipFileMotorLoader.java:56)
	at net.sf.openrocket.file.motor.GeneralMotorLoader.load(GeneralMotorLoader.java:36)
	at net.sf.openrocket.database.MotorDatabaseLoader.loadFile(MotorDatabaseLoader.java:98)
	at net.sf.openrocket.database.MotorDatabaseLoader.loadDatabase(MotorDatabaseLoader.java:67)
	at net.sf.openrocket.database.AsynchronousDatabaseLoader.doLoad(AsynchronousDatabaseLoader.java:103)
	at net.sf.openrocket.database.AsynchronousDatabaseLoader.access$100(AsynchronousDatabaseLoader.java:8)
	at net.sf.openrocket.database.AsynchronousDatabaseLoader$LoadingThread.run(AsynchronousDatabaseLoader.java:126)

---------- Thread information ----------
Thread[DatabaseLoadingThread,1,main]

---------- System information ----------
OpenRocket version: 15.03
OpenRocket source: default
OpenRocket location: C:\Users\Ilgar\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
JOGL version: 2.1.5
Current default locale: tr_TR
System properties:
  app.preferences.id=net/sf/openrocket/startup
  awt.toolkit=sun.awt.windows.WToolkit
  file.encoding=Cp1254
  file.encoding.pkg=sun.io
  file.separator=\
  java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
  java.awt.printerjob=sun.awt.windows.WPrinterJob
  java.class.path=C:\Users\Ilgar\AppData\Local\OpenRocket\app\OpenRocket-15.03.jar
  java.class.version=52.0
  java.endorsed.dirs=C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\endorsed
  java.ext.dirs=C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\ext;C:\Windows\Sun\Java\lib\ext
  java.home=C:\Users\Ilgar\AppData\Local\OpenRocket\runtime
  java.io.tmpdir=C:\Users\Ilgar\AppData\Local\Temp\
  java.launcher.path=C:\Users\Ilgar\AppData\Local\OpenRocket
  java.library.path=C:\Users\Ilgar\AppData\Local\OpenRocket\app;C:\Users\Ilgar\AppData\Local\OpenRocket
  java.runtime.name=Java(TM) SE Runtime Environment
  java.runtime.version=1.8.0_111-b14
  java.specification.name=Java Platform API Specification
  java.specification.vendor=Oracle Corporation
  java.specification.version=1.8
  java.vendor=Oracle Corporation
  java.vendor.url=http://java.oracle.com/
  java.vendor.url.bug=http://bugreport.sun.com/bugreport/
  java.version=1.8.0_111
  java.vm.info=mixed mode
  java.vm.name=Java HotSpot(TM) 64-Bit Server VM
  java.vm.specification.name=Java Virtual Machine Specification
  java.vm.specification.vendor=Oracle Corporation
  java.vm.specification.version=1.8
  java.vm.vendor=Oracle Corporation
  java.vm.version=25.111-b14
  line.separator=\u000d\u000a
  os.arch=amd64
  os.name=Windows 10
  os.version=10.0
  path.separator=;
  sun.arch.data.model=64
  sun.awt.enableExtraMouseButtons=true
  sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
  sun.boot.class.path=C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\resources.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\rt.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\sunrsasign.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\jsse.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\jce.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\charsets.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\lib\jfr.jar;C:\Users\Ilgar\AppData\Local\OpenRocket\runtime\classes
[... 62 lines cut ...]
   > 	at net.sf.openrocket.database.MotorDatabaseLoader.loadFile(MotorDatabaseLoader.java:98)
   > 	at net.sf.openrocket.database.MotorDatabaseLoader.loadDatabase(MotorDatabaseLoader.java:67)
   > 	at net.sf.openrocket.database.AsynchronousDatabaseLoader.doLoad(AsynchronousDatabaseLoader.java:103)
   > 	at net.sf.openrocket.database.AsynchronousDatabaseLoader.access$100(AsynchronousDatabaseLoader.java:8)
   > 	at net.sf.openrocket.database.AsynchronousDatabaseLoader$LoadingThread.run(AsynchronousDatabaseLoader.java:126)
  40      1,119 STDERR (SwingExceptionHandler.java:49) java.lang.IllegalArgumentException: MALFORMED
  41      1,119 STDERR (SwingExceptionHandler.java:49) 	at java.util.zip.ZipCoder.toString(ZipCoder.java:58)
  42      1,119 STDERR (SwingExceptionHandler.java:49) 	at java.util.zip.ZipInputStream.readLOC(ZipInputStream.java:300)
  43      1,119 STDERR (SwingExceptionHandler.java:49) 	at java.util.zip.ZipInputStream.getNextEntry(ZipInputStream.java:122)
  44      1,119 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.file.motor.ZipFileMotorLoader.load(ZipFileMotorLoader.java:56)
  45      1,119 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.file.motor.GeneralMotorLoader.load(GeneralMotorLoader.java:36)
  46      1,119 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.database.MotorDatabaseLoader.loadFile(MotorDatabaseLoader.java:98)
  47      1,119 STDERR (SwingExceptionHandler.java:49) 	at net.sf.openrocket.database.MotorDatabaseLoader.loadDatabase(MotorDatabaseLoader.java:67)
[... cut ...]
  51      1,119 INFO   (SwingExceptionHandler.java:64) Exception handler not on EDT, invoking dialog on EDT
  52      1,136 INFO   (SwingExceptionHandler.java:201) Showing Exception dialog
  53      1,205 DEBUG  (UpdateInfoRetriever.java:173) Update response code: 404
  54      1,205 WARN   (UpdateInfoRetriever.java:184) Unknown server response code: 404
  55      2,205 USER   (SwingExceptionHandler.java:220) User requested sending bug report
  56      2,205 DEBUG  (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

`

JoePfeiffer, 2022-04-17:
Yes, please add the bug report.  I'm having a hard time duplicating the problem; I've now tried your motor file under both Linux and Windows and it's worked fine with both.

SiboVG, 2022-06-09:
@IlgarB could you try to recreate this issue again and send us the bug report so that we can debug the issue?

JoePfeiffer, 2022-08-29:
We couldn't duplicate and no response from user after four months.

closed 2022-08-29 by JoePfeiffer