openrocket/openrocket #1304: OpenRocket hangs on bad thrustcurve

JoePfeiffer, 2022-04-18:
If a motor with a bad thrustcurve is imported into OR, a message dialog is popped up describing the error.  Unfortunately, since the motor loading progress dialog is modal, this message is unresponsive and OR hangs.  Screenshot showing the situation is at https://www.rocketryforum.com/attachments/or_error-png.514239/

JoePfeiffer, 2022-04-19:
Sibo, I'm thinking you're better at this stuff than I am.

SiboVG, 2022-04-19:
Can't really test it myself because it works fine on my machine, but @JoePfeiffer could you verify if [this JAR](https://drive.google.com/file/d/1c5LRrqLwS9QC3_Bhpb3IwOslcOmpMmGr/view?usp=sharing) fixes anything?

JoePfeiffer, 2022-04-19:
Looks like that got it!  Something I probably should have put into the original report:  would it be reasonable to put the full path to the thrustcurve causing the problem in the error dialog, instead of just the filename?  We've had at least one user who installed a bad thrustcurve at some point in the past, forgot about it, and then had a hard time finding the file from just the filename.

SiboVG, 2022-04-20:
> Looks like that got it! Something I probably should have put into the original report: would it be reasonable to put the full path to the thrustcurve causing the problem in the error dialog, instead [...]

Added full filename support in #1309

PR #1309 (2022-04-20): fixes #1304 Improve thrust curve import dialog
This PR fixes #1304 where the thrust curve error dialog was not accessible due to the loading motor dialog's modality. Additionally the full file path of the erroneous thrustcurve is now shown instead of only the thrustcurve name
commit 2c4ebaba (2022-04-20): Merge pull request #1309 from SiboVG/issue-1304

fixes #1304 Improve thrust curve import dialog
closed 2022-04-20 by SiboVG