openrocket/openrocket #1315: [Bug} Changes to parachute automatic resizing feature related to parachute length

hcraigmiller, 2022-04-28:
The automatic parachute sizing feature is a wonderful improvement, and there are three changes that will make it even better.

Currently, the parachute automatic resizing has three problem areas:
1. With the automatic box checked, when a user moves a parachute from a tube having a diameter of A to a tube having a diameter of B (or changes a tube's inner diameter), _the length of the parachute is not automatically resized_; and
2. With the automatic box checked, when a user unchecks the automatic box, _the length of the parachute is not automatically resized to its preset length_. For example, if a preset parachute with a diameter of 2 and length of 3 is placed inside of a tube with an inner diameter of 1, then the parachute will be given an automatic diameter of 1 and length of 12. If the user then unchecks the automatic box, the parachute diameter returns to the preset of 2, but the length stays at 12, rather than also returning to the preset of 3.

Those two changes would be very beneficial. It's already a thing of beauty

hcraigmiller, 2022-04-28:
I've been mulling this around too, and this is what I came up with. The **overriding purpose** of using the manufacturer volume (length and diameter) is to ensure that the selected parachute properly fits in the internal space. So, I believe that _as a "preset" parachute moves from one inner diameter to another, **both** the packed diameter and length should resize_, maintaining the manufacturer volume.

I believe that your concern about user changes to the parachute packed length is already partially addressed where, with the automatic box checked, the user can change the packed length without the automatic box unchecking, and would be fully addressed **if the _user changed packed length_ is used to calculate a _new user volume_** (user length *((PI() * (auto-diameter/2)^2), **and the packed length is thereafter calculated using the new user volume**, as inner diameters change.

The following may already be true _after implementation of the above_, but should be the case:

If the user manually unchecks the automatic box, then the manufacturer volume (both length and diameter) should be restored; currently only the diameter is restored,. 

If the automatic box unchecks because the user changes the diameter, then, as is currently the case, the packed length should be the length at the time the box is unchecked. And, if the user re-checks the automatic box, the diameter and length at that time should be used to compute the volume from which the diameter and length will thereafter be derived as inner diameters change. 

To restore the manufacturer volume after that, the user would simply re-select that same or a different preset parachute.

neilweinstock, 2022-05-06:
I have read this several times and am still not quite sure I follow all the nuances.  Where is this functionality right now?  I just did a fresh build of unstable and I don't see it.  I really need to play with this to understand it.  

Based on what's written above, I doubt there is any behavior that will always do exactly what every user wants or expects, unless we add additional controls and options.

There's also some question of what parameters are allowed to be changed while still maintaining the component's identity as a preset.  Should a user be able to change the volume of a preset chute, where the packed volume is spec'ed by the manufacturer, and still call it the same preset?  I guess some folks can pack a particular chute tighter or looser, but it still seems a bit dicey to me.

SiboVG, 2022-05-06:
> I have read this several times and am still not quite sure I follow all the nuances. Where is this functionality right now? I just did a fresh build of unstable and I don't see it. I really need to play with this to understand it.

There now is an 'Auto' checkbox for the packed diameter of all mass components (parachute, streamer, shock cord, mass component) that, when checked, sets the packed diameter equal to the inside diameter of its parent.
[image]

neilweinstock, 2022-05-24:
Was this fixed in beta3?

hcraigmiller, 2022-05-24:
No.

PR #1604 (2022-08-28): [#1315] Update auto length in MassObject
This PR fixes #1315. New behavior for all mass objects:

- If the 'Auto' checkbox at (packed) diameter is unchecked:
     - No special behavior; you can just edit the diameter and length as you please
- If you then enable the 'Auto' checkbox:
    - The diameter will change to fit the parent component
    - The length will change in such a way that the (packed) volume after enabling the 'Auto' checkbox is the same as the (packed) volume before enabling the checkbox. E.g. if the diameter is 5 cm before, length 5 cm (volume of 98.17 cm^3), enabling the 'Au [... cut]
closed 2022-08-28 by JoePfeiffer