openrocket/openrocket #1319: Feature request: build "base drag hack" into OR

neilweinstock, 2022-04-30:
Some folks like to use the "base drag hack" when simulating short/fat rockets: adding a cone to the back of the rocket, equal in diameter to the aft end of the rocket and diameter * PI long.  This hack produces a better CP calculation, so it's good for evaluating stability, but it's ugly to look at and also must be removed for flight simulation.  This is a pain and also trips up a lot of users.

My request is for OR to have the hack "built-in" in some way.  We'd like it to be incorporated in CP calculation but ignored in flight simulation, and preferably not shown in the rocket view.  I can think of several ways to do it, not yet sure which is best.

1) Have the program automatically determine when to apply the hack, and do it invisibly when calculating CP.  This would require a global preference to enable this functionality, and it should probably provide a visual indicator when the hack is in play.

2) Have a checkbox somewhere to turn on the base drag hack for a particular rocket.  In this case, OR would apply the hack to the aft-most airframe component.  This is very clean and easy to use, but also not very flexible.  There are times when it might be appropriate to apply the hack to something other than the aft airframe component.

3) Have a checkbox in a particular *component* to include the base drag hack for that component.  This is the most flexible implementation, but I would worry about this checkbox being hidden in a component and easy to lose track of.  It's possible that providing a good visual indication when the hack is enabled would solve this, but I'm not sure what that visual indication should be.

4) Have a base drag component that the user could add as a child of any airframe component.  This would contribute to CP calculation but be excluded from flight simulation.  Because it would have its own entry in the component tree, it would be pretty reliably visible.

5) Other than for implementation (1), it would be very nice if the program could detect when it is appropriate to use the base drag hack, and put up a warning for the user or some other indicator.

It'll take some thinking to do this well, but really any implementation would be very welcome, and save users from having to do this all manually.

JoePfeiffer, 2022-04-30:
I'd much rather just simulate off-center base drag correctly, eliminating the need for the hack. I think it'll work to simulate off-center thrust at the same time (see #1062).

At present, drag simply pulls back on the rocket and thrust simply pushes forward.  What's needed is to consider drag and thrust as vectors, with an origin that may not be on the centerline of the rocket.  I suspect this will require lots of changes throughout the code, so I don't want to think about it until after final 22 release.

neilweinstock, 2022-04-30:
I think we're talking about different things; what I'm talking about has nothing to do with off-center drag.  It's a hack purely to correct the CP calculation for short stubby rockets.

In any case, it is surely something to be considered post-22, as always I just want to record it here while it's in my head.

JoePfeiffer, 2022-04-30:
They're the same thing.  My understanding is that the reason CP calculations aren't accurate for short stubby rockets is that when they pivot so they have a non-zero angle of attack their base goes off center, so their base drag is off-center. This is also an inaccuracy for more normal rockets, but it becomes more pronounced as the rocket gets stubbier.

neilweinstock, 2022-04-30:
Yes but the correction hack doesn't require any new off-center drag calculation code, just the inclusion of an additional phantom cone when calculating CP.

Of course, if the CP calculations could incorporate base drag "properly", without the hack, that would be great... but that sounds like a more difficult, longer-term project.  A case of "perfect is the enemy of good", perhaps.

Anyway, something to be discussed and/or debated post-22.

JoePfeiffer, 2022-04-30:
Point taken -- we don't want another seven year release cycle!

I think we're in agreement that we want to handle short stubby rockets better, and the route to take to get there is after this release.

neilweinstock, 2022-04-30:
Bingo. 😀

Charles-McG, 2022-07-06:
My understanding is that the drag property being 'hacked' isn't off-center (as is off axis) but shifting the position along the long axis to a spot behind the base of the frame.  That might be easier to work into the code than fully generalizing to handle radial offsets, too.

neilweinstock, 2022-07-06:
Correct.  A fully generalized solution is obviously desirable, but until that is available embedding the hack into the program seems relatively straightforward.

neilweinstock, 2023-05-23:
Yet another problem with manual base drag hack: screws up the "% of length" stability calculation.

I continue to think that, in the absence of a fully generalized solution, a special "base drag cone" component should be provided that addresses all the manual settings that users must do today.

neilweinstock, 2024-07-18:
Based on extensive discussion of the base drag hack on TRF, I am closing this issue.  There are too many questions regarding the validity of the hack to justify building it into OR.  Users who want to do it can just do it manually.

closed 2024-07-18 by neilweinstock