openrocket/openrocket #1321: OpenRocket - v2022 beta 2 - export to RockSim Drops sub-sub assemblies

spigalau, 2022-04-30:
Our existing model has several parts of the model which have 3 levels of componentry  - eg 1) Switch Band which is attached to 2) Tube Coupler and inside that are 3) Other bits (Electronics, Charge-wells, Bulkheads, etc)

**ORK FIle with full tree of components:**
[image]

**Exported to RKT format and re-opening in OR - missing sub-sub level components**
[image]

**ORK file with Electronics component shifted from sub-sub level to sub level**
[image]

**Exported to RKT and re-opened in OR and now the Electronics appear**
[image]

It appears that the Open Rocket export routine is not exporting out any item that is a sub-sub component.

SiboVG, 2022-10-29:
The issue you're describing only applies to tube coupler sub-sub assemblies. RockSim does not support child components to tube couplers, while OpenRocket does. So OpenRocket has to drop those sub-sub assemblies, because RockSim simply does not support it.
[image]

SiboVG, 2022-10-29:
You know what, my apologies, that was a simple answer of me. It would be better for the export to move the tube coupler sub-components outside of the tube coupler, so that the sub-components aren't just thrown away, and RockSim can still read the file.

PR #1779 (2022-11-01): [#1321] Externalize tube coupler children on RockSim export
This PR fixes #1321 and moves tube coupler children outside of the tube coupler during RockSim export.

I also renamed code occurrences of  `Rocksim` to `RockSim` (personal annoyance of mine).
closed 2022-11-01 by SiboVG