openrocket/openrocket #1324: FreeForm Fin weight calculation is different between 1503 and ver 22 beta 2

sjh114961, 2022-05-01:
[L3 v3.zip](https://github.com/openrocket/openrocket/files/8599934/L3.v3.zip)

 There is indeed a difference. and the only difference is in the freeform fins weight. The 1503 version thinks its 33.6 oz and ver 22 thinks it is 40.8 oz. My manual calculation says 36 774 oz.

hcraigmiller, 2022-05-03:
While it is true that 15.03 and the current unstable differ, that is because there was an error in the 15.03 calculation. Here is a spreadsheet showing that the error is in the 15.03 fillet volume computation.

[image]

The difference appears to be the result of the correction of a 15.03 bug, not a bug in the current unstable.

Recommend confirm and close.

sjh114961, 2022-05-03:
Thank you very much for confirm our suspicion. I had someone run the model through RockSim and got the same result.

closed 2022-05-03 by JoePfeiffer