openrocket/openrocket #1325: [Bug} Motor simulations began throwing an exception commencing on 2022-04-27

hcraigmiller, 2022-05-03:
TRF user bad_idea reports simulation results not-a-number (NaN) value, please report a bug, c=(NaN,NaN,NaN,w=NaN). The bug report with the ,ork file is available [here](https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-2-is-now-available.171295/page-12#post-2273298).

The error is repeatable in Windows 10, commencing on 2022-04-27 sometime after the first merge on that date; there were multiple merges on 2022-04-27, so I am unable to me more precise. There was no issue after the 2022-04-26, not until sometime after the first merge on 2022-04-27.

SiboVG, 2022-05-04:
As far as I can tell this happened after PR #1312 got merged

JoePfeiffer, 2022-05-04:
Wow.  I've got *no* idea how anything I did there could have caused NaNs in the motor code. I'lm working on improving the tube fin calculations, so I'll be looking at this too.

JoePfeiffer, 2022-05-04:
The .ork uses a launch lug with an inner radius of 0 to model a rail button; this causes pressure drag calculation to get NaN

JoePfeiffer, 2022-05-04:
I committed my first thought on dealing with this too quickly.  There'll be another later today.

PR #1329 (2022-05-04): Check for inner radius of tube == 0 to avoid NaN in pressure drop
fixes #1325
closed 2022-05-04 by JoePfeiffer