openrocket/openrocket #1336: Exception when moving Photo Studio window to external monitor

OZ1SEJ, 2022-05-09:
---------- Bug report ----------

[Ubuntu 22.04] I was viewing my rocket (https://github.com/OZ1SEJ/Rockets/blob/main/mirage.ork) in Photo Studio on my laptop's main screen, when I decided to move the Photo Studio window to my external monitor by pressing <super> + <rightarrow>.

(Do not modify anything below this line.)
---------- Exception stack trace ----------
com.jogamp.opengl.GLException: Caught IllegalStateException: net.sf.openrocket.gui.figure3d.TextureCache@1857f800 not initialized. on thread AWT-EventQueue-0
at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1327)
at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
at com.jogamp.opengl.awt.GLCanvas$12.run(GLCanvas.java:1442)
at com.jogamp.opengl.Threading.invoke(Threading.java:223)
at com.jogamp.opengl.awt.GLCanvas.display(GLCanvas.java:503)
at com.jogamp.opengl.awt.GLCanvas.paint(GLCanvas.java:557)
at java.desktop/sun.awt.RepaintArea.paintComponent(RepaintArea.java:264)
at java.desktop/sun.awt.X11.XRepaintArea.paintComponent(XRepaintArea.java:66)
at java.desktop/sun.awt.RepaintArea.paint(RepaintArea.java:240)
at java.desktop/sun.awt.X11.XComponentPeer.handleEvent(XComponentPeer.java:555)
at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5072)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.IllegalStateException: net.sf.openrocket.gui.figure3d.TextureCache@1857f800 not initialized.
at net.sf.openrocket.gui.figure3d.TextureCache.getTexture(TextureCache.java:98)
at net.sf.openrocket.gui.figure3d.RealisticRenderer.render(RealisticRenderer.java:126)
at net.sf.openrocket.gui.figure3d.RealisticRenderer.renderComponent(RealisticRenderer.java:102)
at net.sf.openrocket.gui.figure3d.RocketRenderer.renderTree(RocketRenderer.java:202)
at net.sf.openrocket.gui.figure3d.RocketRenderer.render(RocketRenderer.java:170)
at net.sf.openrocket.gui.figure3d.photo.PhotoPanel.draw(PhotoPanel.java:418)
at net.sf.openrocket.gui.figure3d.photo.PhotoPanel.display(PhotoPanel.java:258)
at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
at com.jogamp.opengl.awt.GLCanvas$11.run(GLCanvas.java:1428)
at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
... 28 more

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.01
OpenRocket source: default
OpenRocket location: /home/steen/OpenRocket/OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: da_DK
System properties:
awt.toolkit=sun.awt.X11.XToolkit
exe4j.moduleName=/home/steen/OpenRocket/OpenRocket
file.encoding=UTF-8
file.separator=/
install4j.appDir=/home/steen/OpenRocket/
install4j.exeDir=/home/steen/OpenRocket/
install4j.jvmDir=/home/steen/OpenRocket/jre
install4j.launcherId=59
install4j.swt=false
[... 299 lines cut ...]
2122 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
2123 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
2124 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
2125 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
2126 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
2127 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2128 311.356 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
2129 311.356 STDERR (SwingExceptionHandler.java:49) Caused by: java.lang.IllegalStateException: net.sf.openrocket.gui.figure3d.TextureCache@1857f800 not initialized.
2130 311.356 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.TextureCache.getTexture(TextureCache.java:98)
2131 311.357 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.RealisticRenderer.render(RealisticRenderer.java:126)
2132 311.357 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.RealisticRenderer.renderComponent(RealisticRenderer.java:102)
2133 311.357 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.RocketRenderer.renderTree(RocketRenderer.java:202)
2134 311.357 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.RocketRenderer.render(RocketRenderer.java:170)
2135 311.357 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.photo.PhotoPanel.draw(PhotoPanel.java:418)
2136 311.357 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.figure3d.photo.PhotoPanel.display(PhotoPanel.java:258)
2137 311.357 STDERR (SwingExceptionHandler.java:49) at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
2138 311.357 STDERR (SwingExceptionHandler.java:49) at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
2139 311.357 STDERR (SwingExceptionHandler.java:49) at com.jogamp.opengl.awt.GLCanvas$11.run(GLCanvas.java:1428)
2140 311.357 STDERR (SwingExceptionHandler.java:49) at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
2141 311.357 STDERR (SwingExceptionHandler.java:49) ... 28 more
2142 311.357 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
2143 311.357 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
2144 319.829 USER (SwingExceptionHandler.java:220) User requested sending bug report
2145 319.832 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

hcraigmiller, 2022-07-12:
With all of the changes that have been made, can this issue be reconfirmed? I can no longer replicate this issue.

[Windows 11 Pro; Version 21H2; OS Build 22000.739; Windows Feature Experience Pack 1000.22000.739.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

JoePfeiffer, 2023-02-03:
Yes.  I just had it happen when moving the window from one of my monitors to another (under X using my three monitors as one wide display).  
Same stack trace.

closed 2026-08-24 by SiboVG