openrocket/openrocket #1344: Exit "preset" mode when any component parameter is changed

hcraigmiller, 2022-05-10:
Currently, the following parameter changes do not exit the "preset" mode:
- Nose cone: shape, shape parameter
- Body tube: length
- Transition: shape, shape parameter
- Tube fins: length
- Inner tube: length
- Tube coupler: length
- Centering ring: thickness
- Bulkhead: thickness
- Engine block: length
- Parachute: Cd, line material, line length, packed length, packed diameter
- Streamer: strip area, aspect ratio, Cd, packed length

Should changing these parameters exit the "preset" mode?

SiboVG, 2022-05-11:
> Should changing these parameters exit the "preset" mode?

I would say yes, exit the preset whenever the configuration does not match the preset configuration

JoePfeiffer, 2022-05-11:
We don't want to be too hasty with this.

Taking body tubes specifically, I want to be able to cut a BT-20 body tube to a specified length without losing the information that it is a BT-20.  Likewise, if I squeeze a parachute into a smaller body tube than the manufacturer's suggested packed diameter, it's still the same parachute.

So some of these parameters do call for leaving the preset (if I change the diameter of my BT-20, it is no longer a BT-20), and some override the preset values without losing the preset.

So my first thought is the length of all tubes can be shortened without losing the preset, as can packed parameters.

hcraigmiller, 2022-05-11:
As I see it, the question is primarily whether the change alters the true character of a component, and secondarily whether a user would expect to keep the "preset" designation even after the change.

Using recovery components as an example, any change in the physical attributes of a recovery device (generally the left-hand side of the configuration pane) changes its true character and should result in loss of "preset" status -- for parachutes, the diameter, drag, surface material, line count, line length, and line material).  Changes that are not to physical attributes (generally the right-side of the configuration pane) should not trigger the loss of "preset" status.  And, users would likely not expect "preset" status to be lost by changing packed diameter or length.

Applying that same approach to tubing, as @JoePfeiffer pointed out, shortening a tube does not change its true character, a change of any other physical attribute does (including lengthening the tube).

As to nosecones and transitions, and change in their physical attributes (I would say excluding shoulders) changes their true character. 

Disks can be made smaller, but not larger, and hole in a disk can be made larger, but not smaller, in real life. So, real life changes should not result in a loss of the "preset" status; things that can't be done in real life, like making a disk thicker, should.

So, where do you draw the line. I would suggest that only a change to a fundamental physical characteristic of a component should result in the loss of "preset" status.

hcraigmiller, 2022-05-11:
As I reflect on the list above, I think adding the following to the current list loose "preset" status is warranted:
- Nose cone: shape and shape parameter
- Transition: shape, shape parameter
- Parachute: Cd, line material, line length,
- Streamer: strip area, aspect ratio, Cd

I would leave the rest until after further user comment.

JoePfeiffer, 2022-05-11:
This is getting perilously close to a metaphysical discussion of "substance" and "accidents"!

I'd say almost any physical change to nose cone, transition, and parachute should remove the "preset".  On streamers, the only one that doesn't is length.  On parts that have shoulders (is that just nose cones and transitions?) the shoulder can be shortened without losing preset.

On bulkheads and centering rings, while you can buy one and modify it, I don't think that it's commonly done.  Any modifications to them lose the preset.

It's interesting that @hcraigmiller  noticed that changes on the left side of the configuration panel generally ought to cause a loss of preset, and changes on the right do not.  As we think about this further, we may also want to change what parameters are where so this becomes universally true as a cue to the user.

Note:  let's not delay beta 3 for this.

hcraigmiller, 2022-05-11:
> I'd say almost any physical change to nose cone, transition, and parachute should remove the "preset". On streamers, the only one that doesn't is length. On parts that have shoulders (is that just nose cones and transitions?) the shoulder can be shortened without losing preset.
> 
> On bulkheads and centering rings. . . [a]ny modifications to them lose the preset.

I agree (parachute and streamer packed dimensions should not be preset conditioned attributes).

neilweinstock, 2022-05-24:
The nose cone and transition items were addressed in a recent PR by @SiboVG.

As for the others: in my opinion, any modification that involves changing the length of something should *not* clear the preset.  Nose cone shoulders absolutely can and are cut down by some folks for various reasons.  Any tube can be cut.

Where it gets really squirrelly is if you *lengthen* a parameter.  Should we distinguish between lengthening and shortening?  I would probably just say "no"; simply ignore changes to length.  Gets too crazy otherwise.

I would agree that any change to centering rings and bulkheads clears the preset.

> It's interesting that @hcraigmiller noticed that changes on the left side of the configuration panel generally ought to cause a loss of preset, and changes on the right do not. As we think about [...]

I've been spoiling to do a re-layout of the editing panels since before I joined this team.  My motivation was not separation of items based on whether they cleared the preset... rather, consistency of layout.  I would like to see parameters that are common to many component types have a consistent layout.  It is easier said than done.  Maybe I'll get back to that effort at some point.

I think that distinguishing preset-clearing parameters by layout would likely be difficult and/or undesirable, because it would (I think) directly work against consistency.  A visual indication of some kind might be better.  But, as with most UI issues, mockups and testing will be needed to sort this one out.  In the meantime it will be good just to get the operation correct.

PR #1385 (2022-05-29): [#1344] Clear preset when centering ring/bulkhead length is changed
This PR fixes #1344 by clearing the preset of the centering ring and bulkhead when the length (thickness) changes.
closed 2022-05-29 by SiboVG