openrocket/openrocket #1348: Does not import RKT pods

BigMacDaddy72, 2022-05-11:
I have not tried many file formats but it seems to be well-known that OR does not import pods from RKT files. It would be incredibly valuable if it could parse the data and at least get whatever comparable elements of the pods imported that it can.

For example, this X-15 RKT files does not import: https://www.apogeerockets.com/downloads/rocksim_files/ApogeeX-15.rkt

neilweinstock, 2022-05-12:
LOL, you beat me to it by an hour. :)

neilweinstock, 2022-05-12:
Our big challenge here is going to be that OR only supports pods attached to the airframe, whereas Rocksim supports pods on other components (most importantly, fins).

I have a suspicion that a pretty large proportion of the pods designs that folks are going to want to import from Rocksim are going to be pods-on-fins, and in many cases there isn't going to be a straightforward way to import them.

JoePfeiffer, 2022-05-12:
Ah, OK.  So resolving #1145 is a prerequisite.

neilweinstock, 2022-05-12:
I *think* so.

PR #1795 (2022-11-07): [#1348] Add support for RockSim importing and exporting of Pods
This PR adds the support for importing and exporting pods from and to RockSim. This fixes #1348, although the pods from the design file of #1348 do not get imported, because those use pods on a nose cone, whereas OpenRocket only supports pods on body tubes. Maybe make a new feature request for pods on nose cones and transitions?

"Demo":
OpenRocket Pods example (simplified to use only one pod - as RockSim does, and an offset angle)
[image]

Imported into RockSim:
[image]

RockSim file imported back into OR:
[image]
closed 2022-11-07 by SiboVG