openrocket/openrocket #1354: Export of simulated latitude and longitude needs more decimals

OZ1SEJ, 2022-05-14:
When doing a simulation and afterwards exporting latitude and longitude, only three decimals are exported, corresponding to a rectangular area 111 meters wide. This is a bit too coarse for my use, at least - why not just two more decimals, which would create a box just 1 meter wide.

Also, the text boxes where latitude and longitude are inputted have a strange behaviour (Ubuntu 22.04, v. 22.02.beta.01) where it's possible to paste e.g. a line break into the box.

Feature request: Export directly as KML-file for visualization of 3D flight path in Google Earth.

Feature request: Built-in 2D top-down map-view for easy viewing of flight ground track.

JoePfeiffer, 2022-05-15:
I should mention -- these are all great ideas (the kml export is especially appealing), and we'll need to figure out what's going on with the lat/long entry (do other number entry fields do it?).  But it would be helpful if they had been four issues so they could more easily be tracked.

vvishwa, 2022-05-17:
I see that default unit for latitude and longitude is degree, for each unit there is associated conversion factor. As a result, it comes to same latitude (decimal precision is still an issue)
eg. 0.7853981633831798 all comes to 45 degrees. If unit is kept to radian it should remain precise to original calculated value.

OZ1SEJ, 2022-05-17:
You should be able to convert between radians and degrees without any loss of precision; 0.7853981633831798 isn't the same as 45°, it's the same as 45.00000000000000°. There must be some rounding at play somewhere.

JoePfeiffer, 2022-05-17:
I should have guessed there would be extra complexities! 

Yes, we definitely want to continue to let users select their units, and just changing units for output shouldn't be causing a noticeable loss of precision.

It looks like a method called MathUtil.doubleToString() is used to produce the string representation, and it's rounding to fit things in five characters.  That method seems to be used in the UI for optimization, in CSV export, and in OpenRocketSaver().  I don't see any reason why file output should ever be limiting the precision, so just not calling textToString() there would seem to take care of it.

vvishwa, 2022-05-21:
Some debugging reveals that the call toUnit(double) uses multiplier which is first initialized, probably when simulation result is prepared. The division always gives a number which only seems rounded but in reality not, so here is trace..
[image]

0.7853981633974483/0.017453292519943295 = 45

	/**
	 * Converts from SI units to this unit.  The default implementation simply divides by the
	 * multiplier.
	 * 
	 * @param value  Value in SI unit
	 * @return       Value in these units
	 */
	public double toUnit(double value) {
		return value / multiplier;
	}

I think the original simulation result is in degrees, when conversion happens it uses artificial multiplier to convert from value (again converted) to same. It will be interesting to find out - getting hand a bit wet to begin

JoePfeiffer, 2022-05-24:
See my earlier comment about doubleToString().  That's where the rounding is happening.

Sparrowhawk120, 2022-06-02:
#1037 and #1036 talk about this issue and a solution already

PR #1427 (2022-06-11): [fixes #1354 & #1037] Add support for custom decimal places and exponential notation in simulation exports
This PR fixes #1354 and #1037. I first implemented an increased, static decimal place for the latitude and longitude (for issue #1354) - which is the first commit which you can just ignore - but then found that it wasn't much extra effort to implement #1037 and add support to enter a custom number of decimal places for the simulation export variables. So that solves #1354 automatically, because users can now just choose the decimal places for any variable.

While I was at it, I also added [... cut]
commit c3a3df9d (2022-06-11): Merge pull request #1427 from SiboVG/issue-1354

[fixes #1354 & #1037] Add support for custom decimal places and exponential notation in simulation exports
closed 2022-06-11 by JoePfeiffer