openrocket/openrocket #1360: [Bug] Booster fins are not considered in stability computation

hcraigmiller, 2022-05-15:
I know this is a crude example, but it shows that when the booster is turned on and off, the stability doesn't change. I discovered this when I was reviewing the booster example provided for issue #1211; the model became less stable with the pods, not more.

See attached .ork

[1360.01.01.Booster_Fins.ork.txt](https://github.com/openrocket/openrocket/files/8696064/1360.01.01.Booster_Fins.ork.txt)
.

JoePfeiffer, 2022-05-15:
This looks like it's working as intended -- the CP varies with roll angle, and the value shown is shown is worst case (furthest forward).  When there are only two pods, the worst case CP doesn't change (94.9cm).  When I change the 'Number of copies" to 3 the CP moves back to 103cm, and when I go to 4 it moves to 104.

There has been discussion of trying to display both best and worst case CP; an idea I like is to show xx-yy in the upper right corner where xx is furthest forward and yy is furthest back.  Then as we rotate the side view of the rocket we could move the icon back and forth to show CP at a given roll angle.

hcraigmiller, 2022-05-15:
I this intended with pods as well?
In the attached .orc., if the pod fins are deleted, there is no change in stability.

[1211.01.04.Pod Example.ork.txt](https://github.com/openrocket/openrocket/files/8696095/1211.01.04.Pod.Example.ork.txt)

JoePfeiffer, 2022-05-15:
What else could it do?  You've got a rocket that (without pods) is rotationally symmetric.  Add two fins in a plane and the worst case CP doesn't change.  If I change the number of pods in the pod set (so the fins are now at a 120 angle) stability changes.  If I change the number of fins on a pod to two they're still in a plane so no change; if I go to three the CP moves back.

I'm attaching a modification to the first example you posted; this one just has two fins on the body, two booster pods, and one fin on each pod.  The pods are at a 90 from the fins on the body.  Now if you turn the boosters off the CP goes all the way up into the nose cone!

[2-fins.ork.txt](https://github.com/openrocket/openrocket/files/8696107/2-fins.ork.txt)

hcraigmiller, 2022-05-15:
My question is different... in these two designs, one uses pods and the other simply a second set of fins. Is the stability supposed to be the same or different? Other than the use of the pods for the second set of fins (otherwise identical in both instances), there is no different in the design

[image]
 That's why I'm a little confused; I understand why the boosters would be treated differently.

JoePfeiffer, 2022-05-15:
I *think* (so I'd have to check to be sure) the simulation is smart enough to figure out fin-fin interference when multiple fin sets are on a single body tube.  It isn't able to figure it out (this one I'm sure of) when the fins are on pods that just happen to occupy the same space as the body tube. That would explain the difference here.

hcraigmiller, 2022-05-15:
Got it. Should I explain this in the Wiki, or will it raise more questions than answers?

JoePfeiffer, 2022-05-15:
I would be inclined to leave it.  Getting the CP vs roll to display would help with a lot of cases (like your original question), and things like using pods to put extra fins on a body tube vs. just putting them on the tube is getting pretty esoteric.

hcraigmiller, 2022-05-15:
As discussed above, the observed behavior is normal for boosters in that the booster stability calculation is as heretofore explained by @JoePfeiffer.

closed 2022-05-15 by hcraigmiller