openrocket/openrocket #1364: Exception when opening 3D finished

sjh114961, 2022-05-19:
[FlatCat.zip](https://github.com/openrocket/openrocket/files/8729757/FlatCat.zip)
When I try to use the 3D finished I get this exception.

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.
Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.ArrayIndexOutOfBoundsException: Index -1 out of bounds for length 1 at jogamp.opengl.windows.wgl.awt.WindowsAWTWGLGraphicsConfigurationFactory.chooseGraphicsConfigurationImpl(WindowsAWTWGLGraphicsConfigurationFactory.java:171) at com.jogamp.nativewindow.GraphicsConfigurationFactory.chooseGraphicsConfiguration(GraphicsConfigurationFactory.java:424) at com.jogamp.opengl.awt.GLCanvas.chooseGraphicsConfiguration(GLCanvas.java:1513) at com.jogamp.opengl.awt.GLCanvas.addNotify(GLCanvas.java:609) at java.desktop/java.awt.Container.addNotify(Container.java:2804) at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4839) at java.desktop/java.awt.Container.addImpl(Container.java:1150) at java.desktop/java.awt.Container.add(Container.java:1001) at net.sf.openrocket.gui.scalefigure.RocketPanel.go3D(RocketPanel.java:258) at net.sf.openrocket.gui.scalefigure.RocketPanel.access$800(RocketPanel.java:88) at net.sf.openrocket.gui.scalefigure.RocketPanel$7.setSelectedItem(RocketPanel.java:310) at java.desktop/javax.swing.JComboBox.setSelectedItem(JComboBox.java:609) at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:654) at java.desktop/javax.swing.plaf.basic.BasicComboPopup$Handler.mouseReleased(BasicComboPopup.java:946) at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:298) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389) at java.desktop/javax.swing.plaf.basic.BasicComboPopup$1.processMouseEvent(BasicComboPopup.java:551) at java.desktop/java.awt.Component.processEvent(Component.java:6391) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.03
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=61.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\sj_h1\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\ANT\apache-ant-1.9.14-bin\bin;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Users\sj_h1\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\sj_h1\AppData\Local\Programs\Python\Python39\;C:\Users\sj_h1\AppData\Local\Microsoft\WindowsApps;;C:\Users\sj_h1\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\sj_h1\.dotnet\tools
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17+35-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17
java.version.date=2021-09-14
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17+35-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln8836521289921952596
line.separator=\u000d\u000a
native.encoding=Cp1252
os.arch=amd64
os.name=Windows 10
[... 213 lines cut ...]
55295 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
55296 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
55297 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
55298 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
55299 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
55300 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
55301 1122.168 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
55302 1122.168 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
55303 1122.168 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
55304 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
55305 1122.168 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
55306 1122.169 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
[... cut ...]
55314 1122.169 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
55315 1122.169 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
55316 1122.169 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
55317 1146.660 USER (SwingExceptionHandler.java:220) User requested sending bug report
55318 1146.660 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

sjh114961, 2022-05-19:
Can't find a link to the beta 2

sjh114961, 2022-05-19:
If this helps 

OS Name	Microsoft Windows 10 Pro
Version	10.0.19044 Build 19044
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	DESKTOP-TSDBAV8
System Manufacturer	Hewlett-Packard
System Model	HP EliteDesk 800 G1 SFF
System Type	x64-based PC
System SKU	F8T90UC#ABA
Processor	Intel(R) Core(TM) i5-4570 CPU @ 3.20GHz, 3201 Mhz, 4 Core(s), 4 Logical Processor(s)
BIOS Version/Date	Hewlett-Packard L01 v02.71, 5/9/2017
SMBIOS Version	2.7
Embedded Controller Version	255.255
BIOS Mode	Legacy
BaseBoard Manufacturer	Hewlett-Packard
BaseBoard Product	1998
BaseBoard Version	Not Available
Platform Role	Desktop
Secure Boot State	Unsupported
PCR7 Configuration	Binding Not Possible
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume1
Locale	United States
Hardware Abstraction Layer	Version = "10.0.19041.1566"
User Name	DESKTOP-TSDBAV8\sj_h1
Time Zone	Central Daylight Time
Installed Physical Memory (RAM)	20.0 GB
Total Physical Memory	19.9 GB
Available Physical Memory	14.5 GB
Total Virtual Memory	22.9 GB
Available Virtual Memory	17.1 GB
Page File Space	3.00 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
Device Encryption Support	Reasons for failed automatic device encryption: TPM is not usable, PCR7 binding is not supported, Hardware Security Test Interface failed and device is not Modern Standby, Un-allowed DMA capable bus/device(s) detected, TPM is not usable
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	No
Hyper-V - Data Execution Protection	Yes

SiboVG, 2022-05-19:
There is a user [here](https://stackoverflow.com/a/69830740/11965105) suggesting he "fixed" the issue by switching back to Java 11. The beta 3 packaged programs now run on Java 17 instead of Java 11, maybe that could be the issue?

hcraigmiller, 2022-05-19:
Unable to replicate issue.

Java version "17.0.3.1" 2022-04-22 LTS
Java(TM) SE Runtime Environment (build 17.0.3.1+2-LTS-6)
Java HotSpot(TM) 64-Bit Server VM (build 17.0.3.1+2-LTS-6, mixed mode, sharing)

[Windows 11 Pro, Version 21H2, OS Build 22000.675, Windows Feature Experience Pack 1000.22000.675.0]

JoePfeiffer, 2022-05-19:
I hate Heisenbugs

sjh114961, 2022-05-19:
I get the same exception when I go into Photo Studio. I would be glad to try the Java 11 thing if someone would lead me through it.

SiboVG, 2022-05-19:
> I get the same exception when I go into Photo Studio. I would be glad to try the Java 11 thing if someone would lead me through it.

Yeah could you download the JAR file for OpenRocket 22.02.beta.03, open Windows Console and type "java -jar {location to the JAR file}" and then hit enter? (You normally can drag-and-drop the JAR file to your console instead of typing out the full path location)

SiboVG, 2022-05-19:
> I hate Heisenbugs

Thank you, another one for my programmer jargon!

hcraigmiller, 2022-05-19:
As to the new PhotoStudio exception being described, unable to replicate the issue.

There was one user during beta 2 testing that stated that he was having similar problems, and that, upon his changing his graphics card settings, the issue went away; I am unsure whether that meant merely updating his graphics drivers, or otherwise. Has looking at the issue from tyhis standpoint been considered?

Java version "17.0.3.1" 2022-04-22 LTS
Java(TM) SE Runtime Environment (build 17.0.3.1+2-LTS-6)
Java HotSpot(TM) 64-Bit Server VM (build 17.0.3.1+2-LTS-6, mixed mode, sharing)

[Windows 11 Pro, Version 21H2, OS Build 22000.675, Windows Feature Experience Pack 1000.22000.675.0]

sjh114961, 2022-05-19:
Support for my built-in graphics system is no longer supported and I have the last update available,

sjh114961, 2022-05-21:
Ran the Beta 3 with Java 11 and it ran just fine. No issue.

SiboVG, 2022-05-22:
> Ran the Beta 3 with Java 11 and it ran just fine. No issue.

Thanks for the test! As expected, it's a Java 17 issue then...

SiboVG, 2022-05-22:
@sjh114961 does running [this JAR](https://drive.google.com/file/d/1S3KMyKXGr3iiJLncOIr3hx0V1JwmDfbJ/view?usp=sharing) with Java 17 solve anything?

sjh114961, 2022-05-23:
No, got same exception

SiboVG, 2022-06-18:
Since we switched the installers back to Java 11, I'm gonna close this issue (the JARs can still run on Java 17, but if JAR-users reports this bug, I'll just happily guide them to this comment to tell them: please run your JAR with Java 11, this will fix it :) )

sjh114961, 2022-06-18:
I do not agree to this closure. The bug exists even though there is a work around.

SiboVG, 2022-06-18:
I do not agree with you disagreeing 🙃. Running OR with Java 11 isn't really a workaround. First of all we very sturdily promote not running OR directly from the JAR, but instead running it from the packaged installers (which are now running back on Java 11). Secondly, we mention in the release notes that running it with Java 17 can result in bugs that are not present when running it from Java 11. And lastly, since the 3D visualization code is in the process of being replaced, this bug is temporally. We simply cannot fix this bug with the old library that OR currently uses for the visualization (believe me, we've tried... we even tried modifying the library's code but were even unable to compile the library because of its archaic code and structure...).

So just in short: this bug should not be produced for users who run OR in a normal way (packaged installer or JAR with Java 11), so in that regard this issue should remain closed.

JoePfeiffer, 2022-06-19:
It does need to remain open until we're able to change 3D libraries since the bug will still be a problem until then.  Yes, we urge people to use an installer, but more people used the jar for beta 3 than Linux.  It did need a more descriptive title, so I've edited that.

SiboVG, 2023-01-13:
A temporary workaround: run the JAR with `java --add-exports=java.base/java.lang=ALL-UNNAMED --add-exports=java.desktop/sun.awt=ALL-UNNAMED --add-exports=java.desktop/sun.java2d=ALL-UNNAMED -jar OpenRocket.jar`.

PR #2232 (2023-06-26): Bump JOGL to 2.5
This PR updates JOGL to the latest snapshot, of version 2.5 (gluegen 961, jogl 1529 found [here](https://jogamp.org/deployment/archive/master/)). This fixes #1364.

@JoePfeiffer could you check whether this causes a regression of #1156?
closed 2023-06-26 by SiboVG