openrocket/openrocket #1382: Feature request: update zoom button icons

SiboVG, 2022-05-26:
The current zoom button icons are honestly unreadable... The actual +/-/frame icon is just too small to know what is what

example from the simulation panel that proves my point (had to hold my laptop 10 cm from my face to actually know which is which):
[image]

SiboVG, 2022-05-26:
I think that a general revamp and modernization of all the pictograms in OR is not a bad idea. @neilweinstock is this something you would be willing to look into?

neilweinstock, 2022-05-26:
I agree about the zoom button icons.  I can fix those easily enough.

As for the rest of the program, though... I actually can't find too many other pictograms in the rest of the program.  Did you have any in mind?

SiboVG, 2022-05-26:
> I agree about the zoom button icons. I can fix those easily enough.
> 
> As for the rest of the program, though... I actually can't find too many other pictograms in the rest of the program. Did you have any in mind?

For example these icons are just all too detailed and hardly distinguishable.
[image]

Also the simulation icons IMO don't need the bevel emboss and drop shadow etc. Just looks a bit outdated.
[image]

neilweinstock, 2022-05-26:
I don't disagree.  Redoing all the icons in the menus is a bit of a job, best accomplished by taking a pre-existing set (free or purchased).

My suggestion for now would be to incrementally fix the worst of them (I agree with the zoom icons and the dots, less inclined to go after the menus right now).

Here are some better zoom in and zoom out icons.  I confess I have little idea yet what to do with the one in the middle.
[image]

SiboVG, 2022-05-26:
I would suggest doing something like this for the middle one, so two brackets [ ]
[image]

And I would indeed use existing icons, for example the ones from Font Awesome or any other open source icon pack (there are a bunch out there).

neilweinstock, 2022-05-26:
Not sure about the brackets.  Others improved slightly.

[image]

SiboVG, 2022-05-26:
Maybe make the brackets slightly smaller so you can separate them more?

neilweinstock, 2022-05-26:
Tried but with no success.  Can fiddle some more.  Make sure you're viewing the latest I put there at 100% size.

SiboVG, 2022-05-27:
That looks really good!

Here's wider-spaced brackets, for comparison.
[image]<https://user-images.githubusercontent.com/243872/170712850-f0422425-3572-4359-9bec-319b7612214f.png>

SiboVG, 2022-05-29:
Fixed in #1386

closed 2022-05-29 by SiboVG