openrocket/openrocket #1392: 22.02.beta.03, MacOS 12.2.1 : "Can't be opened because Apple cannot check it for malicious software"

broofa, 2022-05-30:
I just installed the latest beta, `22.02.beta.03` and upon opening it, I get the following error:

[image]

Note: [There is a workaround](https://support.apple.com/guide/mac-help/apple-cant-check-app-for-malicious-software-mchleab3a043/mac), but it looks like this is an issue that should be addressed here, hence filing this issue.

JoePfeiffer, 2022-05-30:
The workaround is in the Mac instructions on the download page.  The "real" issue appears as issue #1144

hcraigmiller, 2022-05-30:
Have you tries following the instructions here: https://openrocket.info/downloads.html?vers=22.02.beta.03

"Show macOS installation instructions"

closed 2022-05-30 by JoePfeiffer

SiboVG, 2022-05-30:
Also want to point out that we are working on getting OpenRocket code signed by Apple so that it is included as a trusted software on macOS.

broofa, 2022-05-31:
@JoePfeiffer, @hcraigmiller :  FWIW, I ran into this issue after launching `beta.02` and being instructed to "Install Update" (below).  Clicking that button triggers a direct browser download of the `.dmg` image.  There's no warning about this issue in that workflow.

[image]

SiboVG, 2022-05-31:
> @JoePfeiffer, @hcraigmiller : FWIW, I ran into this issue after launching `beta.02` and being instructed to "Install Update" (below). Clicking that button triggers a direct browser download of the `.dmg` image. There's no warning about this issue in that workflow.

The direct download in the software updater is meant to ease the pain of installing a new update. But again: this issue is only temporarily, once we get the code signing to work, the warnings will be gone. Thanks for bringing it up though.