openrocket/openrocket #1396: Supper buggy UI glitches

Reysean05, 2022-06-01:
Hello, ive recently been trying to use open rocket on my windows PC but the UI has been super glitchy making it unusable. it boots up fine but after a few seconds opening any menu causes more to pop out everywhere. I have updated java, updated graphics card drivers, reinstalled, restarted my PC. I have an rtx 3070, 32gb of ram, and a Ryzen 5600x. Please help I need this software.

[image]
.

hcraigmiller, 2022-06-01:
Have you tried this... http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings, This looks like a display setting problem that others experienced

Reysean05, 2022-06-02:
> Have you tried this... http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings, This looks like a display setting problem that others experienced

That seems to of fixed it. Much appreciated.

Reysean05, 2022-06-02:
> Have you tried this... http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings, This looks like a display setting problem that others experienced

That did not work :(. whats strange is it seems to take some time to take effect

SiboVG, 2022-06-08:
Did you try disabling anti-aliasing in the NVIDIA driver?

Reysean05, 2022-06-18:
> Did you try disabling anti-aliasing in the NVIDIA driver?

didnt work

thzero, 2022-06-26:
I've seen that on the current 15 version.   Haven't seen it at all on the betas, or even the old version running under java 11 after ogl updates.

I have same card, but only Intel chip.

This is how I've run the 15.04 version.  Not doing this on the new one, but it may be worth trying.

"C:\Program Files\Eclipse Adoptium\jdk-8.0.312.7-hotspot\bin\java" -Dsun.java2d.d3d=false -jar OpenRocket-15.04.jar

SiboVG, 2024-08-22:
Duplicate of #1667.

closed 2024-08-22 by SiboVG