openrocket/openrocket #1400: Add laminar boundary-layer drag

Mudrost, 2022-06-02:
According to the documentation, all drag is calculated assuming a fully turbulent boundary-layer for rockets. Although small, I think the laminar region should be investigated; the docs say that "the difference in apogee altitude was less than 5% in with all tested designs", but a 5% difference in apogee seems reasonably large to me - specially when the mission goal is apogee accuracy. 

It also states that Barrowman presents methods to calculate drag for that region, which would be mostly friction drag at the nose cone if I'm not mistaken, so I suppose a quick implementation is possible.

Having said that, more experienced users could chime in and tell us if the difference is indeed negligible.

Mudrost, 2022-06-02:
From Barrowman's thesis:
[image]

At incompressible regions, the friction drag coefficient seems to be exclusively related to Re. Again, from the documentation, the point where the flow transitions can be found setting the Reynolds' number to 5*10^5:
[image]

At the compressible subsonic region, the coefficient remains unchanged (Barrowman):
[image]

And finally, at supersonic regions:
[image]
[image]

I'm not sure about the heat transfer coefficient, but the rest seems doable. This is all for friction drag, of course. I think pressure drag would be negatively affected at the nose cones, since laminar flows have more trouble going around edges.

SiboVG, 2022-06-06:
Thanks for the detailed description! Will probably something we'll implement after the next release