openrocket/openrocket #1405: "Show all compatible" doesn't seem to do anything useful

JoePfeiffer, 2022-06-03:
@neilweinstock and I have been talking about the "Show all compatible" checkbox in the preset database chooser dialog.  Neither of us can figure out anything useful it's supposed to be doing, and whatever it's supposed to do we can't find any sign it actually works.

If nobody can figure out what useful function it's supposed to have it should be removed.  If it is supposed to do something useful it needs to be fixed.

JoePfeiffer, 2022-06-03:
OK, I think I've figured what it does:  it lets you populate selection preset dropdowns in "compatible" component types.  Use case looks like this:

I'm designing a model V2.  I open up the database chooser and select "Show all compatible".  I click on Balsa Machining Service's V2 nose cone and V2 boattail.  Now I go back to the nose cone configuration editor and select the nose cone.  I add a body tube.  I go to add a transition, and discover the V2 boat tail has been added to the presets for transitions, and can add it.

Body tubes aren't in the compatible database elements for nose cones, so I wasn't able to grab the right body tube back when I added the nose cone and  boat tail.  To do that I need to look in the body tube database; doing that with "match forward diameter" (or whatever it's called) makes it easy to find the body tube I need.

I'm going to say this looks like a very counterintuitive way to add related components, and doesn't seem any easier than what seems like a more intuitive approach: when selecting nose cone, add the one we want from the nose cone database.  Go to add body tube, look at body tube database with "match forward" and add the right body tube.  Go to add transition, again look at database with "match forward" set and see the V2 transition.

Unless someone wants to make a case for keeping it, I think it should go away.

neilweinstock, 2022-06-25:
I propose to remove this feature.  Even though it has a noble intent, it seems like no one uses it or even understands it.

If many people complain we can always put it back. :)

PR #1580 (2022-08-19): [#1405] Remove 'Show all compatible' in preset chooser
This PR fixes #1405 and removes the 'Show all compatible' checkbox from the preset chooser dialog.
closed 2022-08-19 by SiboVG