openrocket/openrocket #1429: Feature request: use end of nose-cone shoulder to determine bottom and middle of parent component

questionablerocket, 2022-06-11:
OpenRocket currently uses the base of a nose cone where it meets the body as the bottom of the component.  For solid nose cones and closed hollow cones, most hardware will be attached at the end of the shoulder, so one must either remember or check the length of the shoulder when placing any component there.

I believe it would be preferable if instead the end of the shoulder were used as "bottom of parent component," and if that is done, it probably makes sense to include it in determination of "middle of parent component" too.

Please feel free to close this issue if it's not in keeping with the direction the project developers want to go.

JoePfeiffer, 2022-06-11:
We've looked over the request and talked about it, and we feel that for the majority of users this would be counterintuitive; also, similar considerations could exist for the forward end of motor mounts, the ends of all the tube couplers in a complex rocket, and even the forward end of a particular motor for rockets that anchor recovery hardware to the motor.

We're not going to implement this at this time.

closed 2022-06-11 by JoePfeiffer

questionablerocket, 2022-06-12:
Understood, Joe.  Thank you for considering it.