openrocket/openrocket #1439: Simulation export can take a long time

SiboVG, 2022-06-13:
@Sparrowhawk120 recently made a comment about the simulation export taking a huge time to export (up to 77 seconds...). [See his results](https://github.com/openrocket/openrocket/pull/1427#issuecomment-1153424613). I do not have those problems on my machine (export times < 1 s), but this issue should definitely be addressed.

I believe a possible improvement could be to use OpenCSV's CSV export functionality, which is probably a lot more optimized than our own custom CSV export. Tied to that, I would also update the OpenCSV library.

Sparrowhawk120, 2022-06-13:
So, I seem to have addressed the issue with a Windows Update. Not sure what the root cause was but it now takes around 1 second to do exports at full precision. Sorry about that!

SiboVG, 2022-06-13:
> So, I seem to have addressed the issue with a Windows Update. Not sure what the root cause was but it now takes around 1 second to do exports at full precision. Sorry about that!

Darn Microsoft again making the life of a software developer a nightmare 🫥 But great, I see no reason for this issue to be open then

closed 2022-06-13 by SiboVG