openrocket/openrocket #1440: FinSet friction drag counting only one fin

Mudrost, 2022-06-13:
Fin set friction drag seems to be bugged after the refactor #1280. When changing fin count, the total Fin Set friction drag does not change.

```java
// iterate across component instances
			final ArrayList<InstanceContext> contextList = entry.getValue();
			for(InstanceContext context: contextList ) {
				double componentFrictionCD = calcMap.get(c).calculateFrictionCD(conditions, componentCf, warningSet);

				if (c instanceof SymmetricComponent) {
					SymmetricComponent s = (SymmetricComponent) c;
					bodyFrictionCD += componentFrictionCD;

					final double componentMinX = context.getLocation().x;
					minX = Math.min(minX, componentMinX);

					final double componentMaxX = componentMinX + c.getLength();
					maxX = Math.max(maxX, componentMaxX);

					final double componentMaxR = Math.max(s.getForeRadius(), s.getAftRadius());
					maxR = Math.max(maxR, componentMaxR);
					
				} else {
					otherFrictionCD += componentFrictionCD;
				}

				if (map != null) {
					map.get(c).setFrictionCD(componentFrictionCD);
				}
			}
```
I'm not sure if OR treats each fin as a component, but if it treats them as a FinSet I think this is iterating only once per fin set.

SiboVG, 2022-06-13:
Pinging @JoePfeiffer

JoePfeiffer, 2022-06-13:
Yeah.....  it looks like the component analysis isn't reporting correctly.

As you gathered, the friction drag is being calculated for each fin.  Adding a line of debug after line 395 of BarrowmanCalculator.java
```
			for(InstanceContext context: contextList ) {
				double componentFrictionCD = calcMap.get(c).calculateFrictionCD(conditions, componentCf, warningSet);
				log.debug("component " + c + ", componentFrictionCD " + componentFrictionCD);
```
running with java -Dopenrocket.debug=ALL -jar swing/build/jar/OpenRocket.jar>& debug.log

and simulating the "A simple model rocket" example file, the debug output contains:
```
23923          DEBUG [pool-3-thread-1] n.s.o.a.BarrowmanCalculator - component Nose cone, componentFrictionCD 0.08521593306193374
23924          DEBUG [pool-3-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.05224448713612719
23924          DEBUG [pool-3-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.05224448713612719
23924          DEBUG [pool-3-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.05224448713612719
23924          DEBUG [pool-3-thread-1] n.s.o.a.BarrowmanCalculator - component Body tube, componentFrictionCD 0.3799326318921851
23924          DEBUG [pool-3-thread-1] n.s.o.a.BarrowmanCalculator - component Launch lug, componentFrictionCD 0.0
```
so the finset drag is calculated three times.  If we change to four fins, the debug changes to
```
35582          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Nose cone, componentFrictionCD 0.09417863473436353
35582          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.057739372134803256
35582          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.057739372134803256
35583          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.057739372134803256
35583          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Trapezoidal fin set, componentFrictionCD 0.057739372134803256
35583          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Body tube, componentFrictionCD 0.41989256324441077
35583          DEBUG [pool-1-thread-1] n.s.o.a.BarrowmanCalculator - component Launch lug, componentFrictionCD 0.0
```

PR #1504 (2022-08-02): [#1440] Add all context drags to component drag
This PR fixes #1440 by including all context drags to the component drag (for friction and pressure CD) instead of simply setting the component drag to the last context's drag.

FinSets should now display the correct CD:
[image]
closed 2022-08-02 by JoePfeiffer