openrocket/openrocket #1445: [Bug] Ground-hit velocity inaccurate

hcraigmiller, 2022-06-14:
Will post description and pics in a few minutes.

hcraigmiller, 2022-06-14:
A picture is worth 1000 words...

[image]

[X-15.Rev.006.Post-Rebuild.ork.txt](https://github.com/openrocket/openrocket/files/8895434/X-15.Rev.006.Post-Rebuild.ork.txt)

JoePfeiffer, 2022-06-14:
That's... weird.

JoePfeiffer, 2022-06-14:
Your air pressure is 3 mbar.  I'm not sure how this seems to happen some times; it's like it forgets you wanted it to use the ISA atmosphere model (when I checked that box in the "edit simulations" window, the landing speed for simulation 3 becomes 15.3fps

hcraigmiller, 2022-06-14:
That is really good to hear... better user error than otherwise. I will add this to the Wiki for users to check.

closed 2022-06-14 by hcraigmiller

JoePfeiffer, 2022-06-14:
I've seen enough sims suddenly start coming in ballistic because the atmospheric pressure is near 0 that I'm pretty sure there's a bug someplace.  But I won't be reporting it until I can get some idea of repeatable conditions under which it happens.

hcraigmiller, 2022-06-14:
For future reference, if it helps identify the issue, this design was saved on 03/18/2022 with the L1000 as the selected motor. When opened today, the other three simulations no longer had the atmospheric conditions box checked and were set to 3 mbar.  The L1000 was still checked.