openrocket/openrocket #1461: OpenRocket-22.02.beta.04-macOS.dmg Install Failure

AlbyVA, 2022-06-18:
I'm having an issue where trying to install the package to the Applications Directory
fails with an Error Code -36 about something in the package that cannot be read.

[image]

SiboVG, 2022-06-18:
Could you download the installer again (I updated it), should be fixed now. Have no idea what the issue was, I think install4j generated a corrupt installer because I just did a re-run of the installation and now it seems to work).

AlbyVA, 2022-06-18:
Confirmed. Download and Installation was successful this time around. 
Thanks,

closed 2022-06-18 by AlbyVA