openrocket/openrocket #1465: Fit in side view still broken

sjh114961, 2022-06-19:
Fit in side view still broken. See ork and image

[Mercury.zip](https://github.com/openrocket/openrocket/files/8935898/Mercury.zip)
[image]

hcraigmiller, 2022-06-19:
The centering issues appears to be 2D related...

[image]

It never ceases to amaze me as to what pods are used for...

SiboVG, 2022-06-19:
This issue seems to be caused by pods. For example if you open 'A simple model rocket', delete the nose cone, put a pod set in the body tube, a nose cone in that pod set and move the pod set to the beginning of the body tube, the centering is not done correctly.

hcraigmiller, 2022-06-19:
Agreed... if you rearrange the components in the Mercury.ork so that the red nose cone is the first component then the centering is correct. Just interesting that the 3D displays correctly and the 2D does not.

hcraigmiller, 2022-06-19:
Part of the problem lies in the fact that the designs that are being complained of cannot be constructed in the real world. Using this design, for example, the red tip is constructed using a nose cone, and three body tubes. However, there is no transition between the first small diameter body tube and the second (much larger diameter) body tube, leaving the first hanging in the air without any support.

[image]

But, I do agree that the 2D display should account for the extent of the pod components.

closed 2022-06-20 by SiboVG

sjh114961, 2022-06-20:
I tried to do the fix as described above and could not make it work. Could someone attach the fixed ORK here so I can see what was done.

SiboVG, 2022-06-20:
> I tried to do the fix as described above and could not make it work. Could someone attach the fixed ORK here so I can see what was done.

Changing the ORK will not solve the issue, but I merged a PR yesterday that fixes the issue. If you want to play around with it, [here is the JAR file](https://github.com/openrocket/openrocket/suites/6999543371/artifacts/274446489) with the fix.

sjh114961, 2022-06-20:
Yes, this jar file seems to work correctly.Thanks!

PR #1466 (2022-06-20): [#1465] Include xmin in sideview canvas origin calculation
This PR fixes #1465 by including the rocket's xmin position into its sideview origin calculation. For normal rockets, xmin should equal 0, thus nothing should have changed for them. For rockets where some of its child components reach further to the left than their parent component, xmin becomes smaller than zero, which should then also be included in the origin calculation. The design file from #1465 was an example of the child components reaching further to the left. A body tube contained a pod set that was positioned to the left of  [... cut]