openrocket/openrocket #1468: Feature request: add warning when no motor selected in simulation

SiboVG, 2022-06-21:
When running a sim without a motor, you already get a pop-up error message stating there is no motor selected, but additionally, that warning/error could also be added to the hover dialog of the sim. I would even put a new section 'Errors' above the warnings, like so:
[image]

hcraigmiller, 2022-06-21:
When I don't have a motor in a design that _has only one motor_, the name and configuration have data (configuration stating "No motors") and the rest of the fields are blank. Not sure a second "warning" message is needed.

On the other hand, in multi-stage rockets, a missing motor does not state "No motor" in the place of the missing one (using A simple model rocket for example. you get only "A8-3;;B6-0"), the bubble isn't even red, and _a warning would be helpful_.

The same is true if you have a cluster with a missing motor, _no red bubble_, _no warning that not all of the motors are present_.

JoePfeiffer, 2022-08-24:
Superceded by #1611

closed 2022-08-24 by JoePfeiffer