openrocket/openrocket #1470: Bug: Edit window can go outside of monitor

SiboVG, 2022-06-22:
I've seen a number of Reddit users reporting that the edit window could sometimes go outside of their monitor window ([post 1](https://www.reddit.com/r/rocketry/comments/s39c1k/openrocket_edit_window_does_not_open/?utm_source=share&utm_medium=web2x&context=3), [post 2](https://www.reddit.com/r/rocketry/comments/m1e6tn/cant_edit_parts_in_openrocket/)). Quite a horrible bug...

hcraigmiller, 2022-06-23:
I'm i the process of reconfiguring a system for multiple monitors, but I want to make sure I understand the expected behavior before I begin testing.

Does the following accurately decribe the desired behavior with mulitple monitors (using two monitors as an example):
1. If closed while in full screen on 1, when opened it defaults to full screen on 1; and, if closed while in full screen on 2, when opened it defaults to full screen on 2.
2. If not in full screen on 1 when closed, when opened the window defaults to the size when closed on 1; and, if not in full screen on 2 when closed, when opened the window defaults to the size when closed on 2.
3. If not in full screen, and split between screens when closed, when opened the window defaults to the size when closed on the monitor it was most recently tied to.
4. Sub-windows always open on the same monitor as the main program.
5. If multiple instances are open, the monitor when opened after all instances are closed is the monitor on which the last instance was closed.

Is this what we're trying to achieve?

SiboVG, 2022-06-23:
> Is this what we're trying to achieve?

Your described behavior looks correct to me, but most of them are new feature requests. What should be solved in this issue is 'Check if the component config dialog is on the same monitor as the main window, if not, change its location to the main window location', so basically your option '4'.

closed 2022-07-14 by SiboVG