openrocket/openrocket #1479: Automatically open Preset selection window when creating a new component (for those types that support it)

neilweinstock, 2022-06-25:
I propose that whenever creating a new component, the preset selection window should be opened automatically.

This is not a no-brainer, and I'm not sure I've completely convinced myself of this.  But it's worth consideration as a quality-of-life improvement.  Along with it, it might be necessary to provide a way to override it, perhaps a checkbox in the preset dialog that says "Always open this dialog when creating a new <component type>".  This preference should be separate for each component type, rather than global, and therefore I think it would have to be here rather than in a big mess in Preferences.

To some extent, it undercuts the ability to add a part to the drop-down menu.  But I don't know how many people use that, *and* the way it works isn't super useful (that will be a subject of another feature request).

Anyway, something for consideration.  I'm trying to think of ways to simplify the initial entry of a rocket, and this would be one.

hcraigmiller, 2022-06-25:
I disagree with this change. Unless the user builds primarily (if not exclusively) with preset components, this change will create an additional step just to add a part. This is how Rocksim functions, and (to me) it is a significant annoyance to constantly have to close the preset window before I can configure a part (that in OpenRocket is already using automatic sizing).

neilweinstock, 2022-06-25:
Interesting.... your custom parts include custom tubes and nose cones?  If so, I'm now even more impressed by your builds than I was before (if that's possible). :)

Would a global option to enable/disable this function satisfy you?

I can only speculate, but my guess would be that more people build more of the time with stock parts, and typically go to the preset window to choose.  Right now, I'm working on re-populating my favorites for the preset pull-down menu, but I'm realizing that it only really makes sense for some component types, like body tubes or launch lugs.  Nose cones and centering rings are more likely to be a custom selection each time, unless I want to have a list of 30 favorites for each, which is unworkable.  This is why I thought that having a per-component type option to auto-open the selector would be good... I could disable it for body tubes and launch lugs (where my shortcuts would be most often used) but keep it enabled for nose cones and centering rings etc.

hcraigmiller, 2022-06-25:
A preference checkbox would address my concern; to address your concern the checkbox could also be a grid having a checkbox for each configuration window.

neilweinstock, 2022-06-25:
A grid in preferences could work.  I thought it might be nice to have the checkbox in each preset window, because it's when you're looking at that that you might think of wanting to change the preference.  Maybe the checkboxes could be duplicated in both places?

hcraigmiller, 2022-06-25:
I was thinking of something like this:

[image]

Pick and choose, depending on how good the parts library is. Just a thought...

SiboVG, 2022-06-27:
> I was thinking of something like this:
> 
> [image]
> 
> Pick and choose, depending on how good the parts library is. Just a thought... no aversion to having it in both places, but I think there should be one place to set all of them.

Hm, I'm not such a huge fan of that approach for two reasons:
- Makes me feel like I'm in a cockpit with the vast amount of checkboxes
- This seems too intrusive in the sense that the preferences should IMO be kept as stripped down and general as possible

I would propose to put a checkbox in every component's preset dialog like this:
[image]

That still gives you the possibility to have custom settings for every component type, but is better in two regards:
- It is more hidden and stripped down for people who would not bother to use this feature
- It is located where it would actually have an effect, thus linking action and effect (instead of having it in the preferences tab where new users would not even have a clue on which section of OR the setting has an effect)

neilweinstock, 2022-06-27:
OK in that case we are all in agreement. :)

PR #1811 (2022-11-14): [#1479] Add option to automatically open preset dialog when creating a new component
This PR fixes #1479. There is now an option in the component preset dialog for opening the dialog when creating a new component. The dialog will not automatically open when you have created a component, close the config dialog, and edit the component again. It only works for newly created components. This feature is enabled by default. If you want it disabled by default, let me know.

Demo:

https://user-images.githubusercontent.com/11031519/201204718-232c73f5-e9f0-4736-a122-58e0dc4e6151.mp4
closed 2022-11-14 by JoePfeiffer