openrocket/openrocket #1480: Automatically pre-check "Match aft diameter" and "Match fore diameter" options in preset selection window

neilweinstock, 2022-06-25:
I believe that the *vast* majority of the time, users will want to check the "Match aft diameter" and/or "Match fore diameter" options in the preset selection window, to narrow down the (vast) selection of parts.  We should pre-check these to save the user a step.

hcraigmiller, 2022-06-25:
Forgive me if I am mistaken, but this is the current behavior when adding non-preset components. And, when adding preset components, the match boxes should not be set because the part characteristics are specifically defined.

neilweinstock, 2022-06-25:
I think you're talking about the "automatic" checkbox under OD.  I'm talking about this:
[image]

hcraigmiller, 2022-06-25:
That is correct...

As to the "Match... diameter" boxes shown your screen grab, I have always held a certain level of disdain for this feature because it does not function as the user (meaning me in this case) expects. If I add an Estes PNC-50Y nose cone with an outside diameter of .976 in., then choose preset from database, check the "Match fore diameter" box, and then sort by outer diameter, the results range from .941 to 1.014 in. I think everyone can agree that anything other than .976 **is not a "match"** even though one **might** be able to modify the parts to get the body tube to _work_.

If this feature is changed to show only **true matches**, I agree that checking the "Match... diameter" box makes sense. Otherwise, all you are doing is creating a smaller set of unmatched parts to look through, although I guess only having to look through a smaller set may have some benefit.

neilweinstock, 2022-06-25:
The amount of "slop" allowed by the "Match diameter" option is certainly something that should be looked at; it seems too large right now (when trying to match BT50, you should not see BT51 and BT52 on the list).  However  I do not believe that it needs to be exactly the same down to three decimal places... there are times when slightly mismatched parts might be used.  For instance, if using a heavy-walled BT (BT-50H), then you'll usually need to use a mismatched nose cone because virtually no commercially available cone is designed to match BT-50H.  But that's a much narrower slop than the current filter is allowing.

I don't know how the slop is currently determined, whether it's by percentage or by a fixed amount.  It would probably require some study to determine what a better value might be.

And I would argue that even with the slop, *reducing* the size of the list is still valuable.  It is certainly more useful to have it checked than not, isn't it?

hcraigmiller, 2022-06-25:
Yes, even after sorting by outer diameter, I guess it's easier to locate one out of 50 than one out of 500, for example. As to the "slop" value, anything more than .002 would be of little or no value in real life. The other consideration is that **any difference in the outer diameter** is going to trigger a discontinuity warning, and it is my view that **any discontinuity in the outer diameter should trigger a discontinuity warning**. 

That said, building in "slop" is building in a warning that users already have difficulty understanding; there is even an issue asking that the point of discontinuity be identified.  Is building in "slop" the approach that should be taken? A little wiggle at 200 ft/sec may be okay, a little wiggle at Mach 1.8... not so much.

neilweinstock, 2022-06-25:
> Yes, even after sorting by outer diameter, I guess it's easier to locate one out of 50 than one out of 500, for example. As to the "slop" value, anything more than .002 would be of little or no [...]

Well, the inadequacy of the current warning system is a separate matter, one that I am thinking about quite a bit lately.  However, I understand your concern here. 

I don't think it's really an argument against this feature request, which is to pre-check those options.  I think your beef is with the way those options work.  Is that a reasonable summary?

hcraigmiller, 2022-06-25:
Yes.

JoePfeiffer, 2022-07-11:
It also seems to me that it's a good idea to do this, regardless of possible future refinements to how the checkboxes work.  Over on slack Jonathan Jagt has offered to take it on for his first PR; I suggest we encourage him to do it.

SiboVG, 2022-08-10:
Fixed in #1547

closed 2022-08-10 by SiboVG