openrocket/openrocket #1482: NoCalssDefFound ServiceCOnfigurationError - Adding a Script Extension to the Simulation

thzero, 2022-06-25:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.
Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NoClassDefFoundError: sun/misc/ServiceConfigurationError at net.sf.openrocket.simulation.extension.impl.ScriptingUtil.getLanguages(ScriptingUtil.java:63) at net.sf.openrocket.simulation.extension.impl.ScriptingConfigurator.getConfigurationComponent(ScriptingConfigurator.java:58) at net.sf.openrocket.simulation.extension.impl.ScriptingConfigurator.getConfigurationComponent(ScriptingConfigurator.java:34) at net.sf.openrocket.simulation.extension.AbstractSwingSimulationExtensionConfigurator.configure(AbstractSwingSimulationExtensionConfigurator.java:47) at net.sf.openrocket.gui.simulation.SimulationOptionsPanel$5.actionPerformed(SimulationOptionsPanel.java:236) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:369) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1020) at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1064) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233) at java.desktop/java.awt.Dialog.show(Dialog.java:1070) at java.desktop/java.awt.Component.show(Component.java:1716) at java.desktop/java.awt.Component.setVisible(Component.java:1663) at java.desktop/java.awt.Window.setVisible(Window.java:1031) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005) at net.sf.openrocket.gui.main.SimulationPanel.openDialog(SimulationPanel.java:759) at net.sf.openrocket.gui.main.SimulationPanel.openDialog(SimulationPanel.java:768) at net.sf.openrocket.gui.main.SimulationPanel.access$1100(SimulationPanel.java:74) at net.sf.openrocket.gui.main.SimulationPanel$20.mouseClicked(SimulationPanel.java:478) at java.desktop/java.awt.AWTEventMulticaster.mouseClicked(AWTEventMulticaster.java:278) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6638) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4556) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) Caused by: java.lang.ClassNotFoundException: sun.misc.ServiceConfigurationError at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:471) at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589) at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522) ... 84 more

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02.beta.04
OpenRocket source: default
OpenRocket location: D:\users\thzero\Downloads\OpenRocket-22.02.beta.04.jar
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
file.encoding=Cp1252
file.separator=\
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=D:\users\thzero\Downloads\OpenRocket-22.02.beta.04.jar
java.class.version=55.0
java.home=C:\Program Files\Microsoft\jdk-11.0.12.7-hotspot
java.io.tmpdir=z:\temp\
[... cut ...]
4216 25.531 STDERR (SwingExceptionHandler.java:49) at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
4217 25.531 STDERR (SwingExceptionHandler.java:49) ... 84 more
4218 25.531 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
4219 25.531 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
4220 27.664 USER (SwingExceptionHandler.java:220) User requested sending bug report
4221 27.670 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2022-06-25:
Duplicate of #1270

closed 2022-06-25 by SiboVG

thzero, 2022-06-26:
Argh, thanks.. I'd looked, but since the bug trace is attached, probably why I didnt' see it.