openrocket/openrocket #1485: Allow multiple simulations in same plot

neilweinstock, 2022-06-25:
For comparative purposes, it would sometimes be useful and/or interesting to show multiple simulations in the same plot.  Request by "waltr" on TRF: https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-4-is-now-available.171295/post-2291435

The way it might work is: multi-select simulations, then click plot/export.  When plotting, there should be a list of checkboxes, one for each selected simulation, that lets you toggle its visibility.

It would probably be reasonable to limit the number of sims you could include in the same plot.

I'm not sure how to visually distinguish the different sims within the plot.  Having a completely different set of colors for each is probably not practical (would be impossible to tell what's what); maybe different dash patterns for each?

JoePfeiffer, 2022-06-26:
I can't believe I never thought of this.  It would be *extremely* useful.

SiboVG, 2022-06-27:
Maybe something like this? So as Neil proposed, just use different line types for each sim, and add those line types to the legend, and use the same color for same-typed data. There are also checkbox to enable or disable certain simulations
[image]

I think it would also be helpful to have a hover dialog over the simulation checkboxes that give the most important information about that simulation, e.g. the flight configuration, maybe even some simulation settings like the wind speed?
[image]

neilweinstock, 2022-06-27:
That's pretty close to what I was imagining.  I thought it might be nice if each simulation checkbox were on its own line, with whatever description is necessary coming after it (rather than in tooltips).  That way you could get a screenshot that shows everything, and not be dependent on mousing around.

Other than configuration, I'm not sure what other information to show for each simulation.  In your tooltip there, how did you decide to show wind speed?

SiboVG, 2022-06-27:
> That's pretty close to what I was imagining. I thought it might be nice if each simulation checkbox were on its own line, with whatever description is necessary coming after it (rather than in [...]

Agreed, in that case I would put the checkbox to the right of the plot, because if you'd put a new simulation line (checkbox + description) underneath the plot, the sim window height can grow too large.
[image]

Also not really sure what other information could be useful... The wind speed was just the wind speed set in simulation 2. My idea was that if there are differences between the simulation configurations, those could be displayed. Maybe just list the configuration and if there is a difference between any of the simulation parameters, you could add those details? E.g. if all the simulation settings between sim 2 and 3 are the same, just display
```
Simulation 2
[B4-4]

Simulation 3
[C6-3]
```
But if let's say the wind speed of sim 2 is set to 2 m/s, and for sim 3 to 10 m/s, then display:
```
Simulation 2
[B4-4]
Wind speed: 2 m/s

Simulation 3
[C6-3]
Wind speed: 10 m/s
```

Charles-McG, 2022-06-28:
The simulation name is editable, so the user can embed the important differences in the name of the sim in a way that's parsable to them.   It might be nice to have a function to generate those differences automatically - but as Neil points out - they can pile up fast.  The motor config is the most obvious - but it might not be what the user happens to be studying.