openrocket/openrocket #1486: Software-Breaking Bug

SpaceBoi-Juno, 2022-06-25:
Whenever I open the application, it appears to be working properly. However, whenever I click on a drop box (to switch a nosecone shape to conic, parabolic, or otherwise, as an example), the window pops up again as a graphical glitch, covering up everything behind it. It's hard to explain with words, so I linked a screenshot that shows my issue

[image]
.

This changes whenever I hover over a new clickable button. The issue happens every time I use the installer for both the Beta and the current version, even after redownloading.

SiboVG, 2022-06-25:
Did you try changing your OR settings to use off-screen rendering? http://wiki.openrocket.info/Downloading_%26_Installing#Check_Settings

hcraigmiller, 2022-07-27:
Were you able to resolve this issue using the suggestion of @SiboVG?

SiboVG, 2022-08-14:
Closing due to inactivity.

closed 2022-08-14 by SiboVG