openrocket/openrocket #1488: After selecting a preset component, put focus on most commonly edited value

neilweinstock, 2022-06-25:
This is primarily focused on body tubes, but could apply to other components as well (centering rings, especially)

Tubes of all kinds (body tubes, inner tubes, launch lugs) are usually cut-to-length... or at least, edited-to-length in OR.  Therefore, after selecting a preset body tube, it would a small additional convenience to automatically put focus on the "length" field and select its contents, so you can immediately type in the length, rather than needing to mouse over to the field and double-click.

To me this is a no-brainer for tubes.  For other body components I'm not sure there's a good analog.

For internal components, it's "position" that is going to be edited most of the time, and that's the field that I'd pre-select.

I haven't gone through all the others to make this determination yet.

Basically, this is another way to reduce needless redundant pointing and clicking, to make it quicker and easier to get an initial design into OR.

hcraigmiller, 2022-06-25:
Speaking as to when a preset has been selected, I agree with @neilweinstock that when selecting tubes (body tubes, tube fins, launch lugs, inner tubes, couplers, and engine blocks), the length is the best element to have for an edit default.  And, with regard to centering rings, bulkheads, and all mass components, position is likely the best element to have for an edit default.

When not using presets, I prefer the current behavior.

neilweinstock, 2022-06-25:
Agree completely, this behavior is intended purely for preset selections.

PR #1508 (2022-07-04): [#1488 & #1506] Focus most commonly edited parameter after preset selection + Highlight spinner text after focus gained
This PR fixes #1488 and #1506. If setting focus on any spinner text field (either by clicking on it when it is not yet in focus, or by pressing the 'Tab' key until that object is in focus), all the text in the text field is highlighted/selected (fixing #1506). When you click again in the text field, e.g. the text '123' is highlighted and you click between the 1 and 2, then the highlight will stop and the cursor will be placed between 1 and 2, so '1|23'.

For the following com [... cut]
closed 2022-07-04 by SiboVG