openrocket/openrocket #1503: Feature request: Add Launch Conditions and Simulations Options to Motors & Configuration tab

hcraigmiller, 2022-06-28:
After a somewhat brutal critique on TRF, I think that @neilweinstock may have identified the complaint in a way that it could be understood, which is that there is a disconnect between preparing to launch the rocket (including launch conditions) and changing those settings afterward.

Currently, OR has preset launch conditions that attach to each simulation when that simulation is run, but when configuring a specific flight you cannot see those settings without going to _Edit -> Preferences -> Simulation_ and _Edit -> Preferences -> Launch_. And, the complaint is that this is neither intuitive nor convenient (or as @neilweinstock would say, it is considered an annoyance).

So, this is my suggested change...

1. Leave the global simulation and launch settings in the preferences as currently exists.
2. Leave the _Edit simulation_ functionality on the _Flight simulations_ tab as currently exists.
3. But, add launch and simulation settings tabs to the _Motors & Configuration_ tab so that users can **see** and **change** those settings as desired, without have to go through the preferences to find them.

Something like this:

[image]

Where conditions (launch conditions) and options (simulation options) appear with the other flight configuration counterparts.

This may require some reformatting of the content where pop-up information is being reeassociated/presented with a tab, but that can be figured out. The additional data would also need to be duplicated when a configuration is duplicated.

tjkopena, 2022-06-29:
This seems like a reasonable proposal. But I think the suggested data relocation might actually increase frustration, and I suggest that the true value of the proposal is simply surfacing the launch parameters more prominently & obviously.

Moving the launch parameters out of simulations and into flight configurations is mostly a lateral move of the data, not a fundamental workflow or modeling improvement. It very well might make more sense to more people, but there's a good argument that they make more sense bound with particular simulations (as currently).

I'm inclined to believe that users mostly don't play around much with simulation options proper or plot settings. So if you take the launch parameters out of the simulation objects, then there's greatly reduced utility to simulation objects. Almost nobody is going to have a combined flight+launch configuration (say, B6-4, @ ejection, windy, Maryland) and then want to create multiple simulations of that configuration in order to, say, change the geodetic calculation or plot axes. Sure, some people might want a quick way to produce multiple different plots. But that's got to be a much smaller use case than people using the default plots to look at combinations of flight configurations and launch parameters. So with this change usage will tip even further toward simulations being 1-to-1 with flight configurations in much actual practice, and people will wonder even more why they have to bother creating simulations in the first place---at that point why not just simulate direct from a flight configuration using some global sim settings? In addition, people that want to examine multiple launch conditions, say high and low wind, will have to maintain multiple flight configurations that are otherwise redundant. So, e.g., the rocket view will then list several configurations that are the same thing in that perspective (motor loadout). These people are then also left with a bunch of annoying redundancy.

For what it's worth, if a good-sized change like that is to be made, then I argue that [breaking launch parameters out into their own object/workspace](https://www.rocketryforum.com/threads/configurations-simulations-tweaks-to-make-the-awesome-state-of-or-even-more-awesomer.173437/) is the way to go.

But this proposal above does suggest a very straightforward change that would improve things greatly: Simply merge the Edit Simulation and Plot/Export dialogs into tabs of one single dialog.

Start by combining the two buttons into one Edit/Plot/Export button:
[image]

By clicking that button or double clicking the simulation row, you get a modal combining the current Edit and Plot/Export dialogs:
[image]

Changes:
* All four panels of the two dialogs are in this one modal as tabs.
* The awkward "<< Edit" ">> Plot" navigation buttons are dropped.
* The modal has buttons at the bottom for Plot and Export (these are present on all tabs, as are the name & flight configuration controls).

This seems to have a couple advantages over the current UI:
* There's a very straightforward navigation path through all the editing and viewing.
* It's much more consistent about launch parameters, simulation options, plot, and export options all being a single bundle of one particular analysis view.
* You can quickly tweak launch parameters and bring up a new plot or export of them without clicking to the other tab/dialog.

I think this tweak makes the launch parameters and simulations workflow & screen a lot more clear. And it seems to be a very simple code modification to the two dialogs, much smaller than either proposals above about moving or extracting the launch parameters out of simulations.

tjkopena, 2022-06-29:
It's been ~20 years since I did any Swing programming. But if there's consensus that the merged dialog proposal would be a good change, and nobody else has the capacity to do it quickly, I would make time myself in July to try to implement.

neilweinstock, 2022-06-29:
1) Craig's original proposal up at the top does address the concern I raised on TRF, which was the separation of motor configurations and simulation settings.  Putting the default simulation preferences on the motor config page is kind of unrelated to that.  Plus, I'm not really sure why that would go with motor configurations.  I could see putting those setting on the sim page, maybe, under an "Edit simulation defaults" button or something like that.  Convenient maybe, but not solving the real problem here.

2) Joe's suggestion is for something completely different, and therefore should go in its own issue, or perhaps be discussed in Slack.  I'm receptive to the idea of merging all the settings into a single dialog (launch/sim/plot/export), although from looking at those mockups I'm also not really convinced yet of the implementation.  Certainly, the current organization leaves something to be desired.