openrocket/openrocket #1506: After a single click in an editable field, start by selecting the entire value

neilweinstock, 2022-07-01:
Right now if I click my mouse in an editable field, the first action is for the cursor to appear in front of the existing value:
[image]

This is the wrong behavior most of the time.

Instead, the most useful action upon first click into the field is to select the entire field.
[image]

That way, if the user's desire is just replace the old value with a new one, then can just start typing, without having to execute extra clicks to select the whole value.

If their desire is to edit at one point in the field, then they can just use the arrow keys to locate the correct spot.  They would have had to do that anyway in the existing behavior.

An alternative (inferior, in my opinion, but still better than what's there today) would be to put the cursor in the location where the user clicks, rather than always going to the beginning of the field, which is often the least useful thing to do.

Some of this behavior might be constrained by the Swing, I don't know.  But it sure is annoying the way it is now.

SiboVG, 2022-07-01:
(Assigning myself because I'm working on #1488 and need to do something similar)

neilweinstock, 2022-07-01:
Thank you kindly :)

hcraigmiller, 2022-07-01:
The behavior that most of the programs I work with that use "cell" type information is this...

1. A left-click anywhere in the cell selects all of the cell content, a change in that data of any type erases the existing data and substitutes what is being entered.
2. A second left-click in the cell positions the cursor where the pointer is when the click occurs, and allows data to be added or deleted at that insertion/deletion point. For example, if the field contained 2345, the first left-click selects "2345" and the second left-click, with the pointer between the "3" and the "4", deselects "2345" and puts the cursor between "23" and "45".

As @neilweinstock related, this would likely be the best desired behavior.

closed 2022-07-04 by SiboVG